willprice/python-omxplayer-wrapper #1: Pyomxplayerng Import Error

JonnyAlpha, 2015-01-21:
Hi;

Not sure if this is the right way to report this but I have been trying to use your Python-Omxplayer.
I have never installed from Git Hub before but think I did it OK?

Anyway have tried out the example code in your read me but get an Import Error:

from pyomxplayerng import OMXPlayer
ImportError: No module named pyomxplayerng

Is this an issue or just me?

willprice, 2015-01-21:
Checkout the readme for an example

JonnyAlpha, 2015-01-21:
That;s the code I used?

willprice, 2015-01-21:
My bad, try import from omxplayer instead as I changed the package name. I'll be rebranding it soon enough so people don't confuse it with other python wrappers.

JonnyAlpha, 2015-01-21:
Sorry Will bear with me nube here:
Do you mean instead of:
from pyomxplayerng import OMXPlayer
Try:
from omxplayer import omxplayer

Doesn't make sense?

JonnyAlpha, 2015-01-21:
Here's my code:
#!/usr/bin/env python

from omxplayer import omxplayer
from time import sleep

player = omxplayer('/home/pi/My_Projects/CentralAI/Firefly.mp4')

player.play()
sleep(5)
player.pause()

player.quit()

willprice, 2015-01-21:
Nearly. `from omxplayer import OMXPlayer`, the class name is OMXPlayer and the package name is omxplayer. Let me know if you have any more trouble

JonnyAlpha, 2015-01-21:
from omxplayer import OMXPlayer 
Many thanks Will, I missed the capital P in OMXPlayer, video plays in full screen successfully but produces a whole load of debug lines in the python shell window.

Any idea how I can run the video in a window rather than full screen AKA --win 100 100 500 500 as per running omxplayer via the command line?

JonnyAlpha, 2015-01-21:
Also in your example should the video play for 5 seconds and then quit, seems to continue? I changed the sleep time to 1 and it still plays the whole video which is about 30 seconds long?

willprice, 2015-01-21:
Didn't realise that `omxplayer` had a `--win` option... nice. I'll add that
at some point. Currently it's unsupported.

Can you post the output debug output?

Cheers,

PS. it's all very alpha at the moment, but I've just taken on some work to
make it more robust. After I finish exams I'll have more time to work on it
:)

JonnyAlpha, 2015-01-21:
OK thanks Will, great work, looking forward to updates, just clicked to follow you if that's how I get notified??
Good luck with exams, i'm doing a Degree in Primary Teaching :-(

willprice, 2015-01-21:
I think if you 'watch' the project then you'll get email on new issues, and
updates I upload.

Out of interest, what are you hoping to do?

Cheers

JonnyAlpha, 2015-01-21:
Just created my first fork and pull request for an updated Readme file, gotta love this sharing :-)

JonnyAlpha, 2015-01-21:
Clicked watch, what am I doing - project started trying to create a visual front end for an RPI based robot kind of like Holy from Red Dwarf. However I am now trying to create a tool to assist with teaching or behaviour management but still using a visual font end for a robot. Trying to do it for an assignment due in March :-)

closed 2015-02-07 by willprice