willprice/python-omxplayer-wrapper #3: Multiple instances of omxplayer cause weird display artifacts

willprice, 2015-02-07:
Instantiating multiple omxplayer objects cause weird display artifacts and cause playback problems, explore this and decide how to modify the design to eliminate this issue

willprice, 2015-03-01:
Consider refactoring `OMXPlayer` into a singleton

Depado, 2015-04-20:
Maybe you could use layers to fix this. Depends what you call a weird display artifact. You could set a layer in your OMXPlayer class. (Which would translate to the `--layer` option when starting omx)

willprice, 2015-04-20:
Thanks for the tip, I didn't realise this was a feature. I'm undecided as to whether it's  worth the effort. Where would you want to use this?

Depado, 2015-04-21:
Ha I don't really know. It's just a tip maybe it could fix. 
I'm currently working on the transparency control via dbus. I'll create a pull request once it's done. Though it only works with the development version of omxplayer so I don't know maybe you could create a dev branch or something like that. What do you think ?

closed 2015-06-01 by willprice