willprice/python-omxplayer-wrapper #7: Running two videos consecutively

willprice, 2015-03-01:
Trying to run two videos consecutively doesn't work, the second video does not play.

jthuraisamy, 2015-03-01:
Is it that the second video does not play at all, or it plays simultaneously with the first video?

willprice, 2015-06-05:
This is now fixed and can be verified by running `smoke-tests/test2.py` which runs two consecutive videos

closed 2015-06-05 by willprice

JonnyAlpha, 2015-06-15:
Will, just tried to run the smoke-tests/test2.py, python hung saying that OMXplayer had no attribute exit, so I changed the code to vid1.quit() and vid2.quit(). Was this a typo?

JonnyAlpha, 2015-06-15:
Just played with the code and successfully played two videos (one immediately after the other) with no gaps, a seamless transition which is exactly what I was looking for earlier this year, brilliant, many thanks Will, I can now continue with my project :-)

willprice, 2015-06-19:
Thanks for the correction Jonny, I'll push the edits later today.
:+1:

alexis85120, 2019-08-01:
Hi Will, I don't understand. How do you run smoke-tests/test2.py ? And what will it help to play 2 videos ?