willprice/python-omxplayer-wrapper #8: Consider switching to a permissive licence

jthuraisamy, 2015-03-01:
I just noticed that this project is licensed under the GPL, which may significantly restrict its adoption and use.

This is because any and all other projects which depend on this library would be forced to switch their own licences to GPL and release all of their source code even if its relation to this library is minimal. Were it to be licenced under a MIT-style licence, it would be possible to avoid that.

That said, the decision is yours, of course.

willprice, 2015-03-01:
How about LGPL? I'd rather not go with MIT as it's so permissive and someone could take the library modify it, and not contribute back improvements.

jthuraisamy, 2015-03-01:
I agree.  LGPL would work. :+1:

willprice, 2015-03-01:
Thanks for the tip, I wasn't aware GPL was /that/ restricitve. I'll add a commit

willprice, 2015-03-01:
Fixed in ae7baf3ec1ae0deac93c378c151eb1c3f37f9bf2

closed 2015-03-01 by willprice