willprice/python-omxplayer-wrapper #11: omxplayer launching and autoplaying and not running commands?

lanewinfield, 2015-04-21:
Hey there,

First off, apologies ahead of time—I'm pretty new at this, so some of this might sound dumb.

When I use the example script, none of the additional commands (play, pause, etc.) act when I run it. Instead, omxplayer starts up, autoplays the video I gave it, and it runs indefinitely.

Am I doing something wrong here?

willprice, 2015-06-02:
Hi @lanewinfield, this is probably a (very severe) bug!

I've been pretty busy with exams hence the very late reply, my apologies. I'll be working to try and get stuff working reliably in the next few weeks. I seem to be encountering the same behaviour. If you look inside the `smoke_tests` folder there are some tests you can run that should be fairly self explanatory, the first one just plays and pauses a video, if you observe incorrect behaviour I'd be grateful if you report it.

Give me a few days first as I'm making quite a few changes to the module!

willprice, 2015-06-05:
Have a bash now, I've also uploaded a distribution to pypi, so you should be able to install with `pip install omxplayer-wrapper`

willprice, 2015-06-18:
Closing since there has been no response in 2 weeks, if this is still a problem please reopen the issue

closed 2015-06-18 by willprice

lanewinfield, 2015-06-18:
Thanks, Will. Sorry I haven't had a chance to play with it lately.

Will let you know when/if I get a chance.

willprice, 2015-06-18:
No problem, I'm in no rush either! Cheers