willprice/python-omxplayer-wrapper #15: Provide a callback for when the process exits

willprice, 2015-06-01:


MydKnight135, 2015-06-19:
I need something like this myself to allow for running another video, this time on loop, after the first one finishes.

willprice, 2015-06-19:
@MydKnight135 So you would be opening a new video/new instance of OMXPlayer? 

This clashes with #18, if I implement a daemon mode then there is no need for a callback for process exit, but that would be quite complicated. I may add this as an intermediary solution until I implement a daemon mode, if that is what should be done ;)

willprice, 2015-09-06:
Decided against daemon mode, so it would make sense to implement this.

willprice, 2015-09-06:
This should be quite easy by monitoring when the process exits.

Alternatively we could run with `--loop` if the DBus `seeked` signal is generated which would allow us to pause.

willprice, 2016-10-23:
We spin up a monitor when starting the process, perhaps we can inject an exit callback there that is called once the monitor detects the process exit.

willprice, 2018-11-03:
Present in develop branch as of https://github.com/willprice/python-omxplayer-wrapper/commit/75dbfa2d199256627ebede9ff470b400e51ca003

PR #116 (2017-12-20): Callback when process exits and return exit status
This should fix #15 and #106. Basically I set up an event keeping it consistent with previous events.
This way in the main script you can run:
`player.exitEvent += lambda _, exit_code: print("OK") if exit_code  == 0 else print("NOT OK")`

or just

`player.exitEvent += lambda _, _: print("OMXPLAYER TERMINATED")`
closed 2018-11-03 by willprice