willprice/python-omxplayer-wrapper #16: After player.play() ends playing video player.playback_status() = None

eN0Rm, 2015-06-10:
```
#!/usr/bin/env python
# -*- coding: iso-8859-15 -*-
# vim: tabstop=8 expandtab shiftwidth=4 softtabstop=4

import RPi.GPIO as GPIO
import time
from omxplayer import OMXPlayer

GPIO.setmode(GPIO.BCM)
PIR_PIN = 18
GPIO.setup(PIR_PIN, GPIO.IN)

def MOTION(PIR_PIN):
    if not player.playback_status() == "Playing":
        print "Starting Video"
        player.play()
        print player.playback_status() 
    else:
        print "The video is playing"

print "Expology Simple Player (CTRL+C to exit)"
print "Loading video"
player = OMXPlayer('/home/pi/film.mp4')
print "Video loaded"
print "Ready"
print "Waiting for PIR sensor"
print player.is_playing()

try:
    GPIO.add_event_detect(PIR_PIN, GPIO.RISING, callback=MOTION)
    while 1:
        time.sleep(100)
        #print "I've slept"
except KeyboardInterrupt:
    print "Quit"

    player.stop()
    player.quit()
    GPIO.cleanup()
```

willprice, 2015-06-18:
Could you provide a minimal example of how to reproduce the unexpected behaviour please? Once provided I'll take a look into it ;)

willprice, 2016-10-08:
Seeing as there has been no response I'm assuming the issue can't be replicated.

If someone can give me a minimal example demonstrating the problem I'll happily investigate.

closed 2016-10-08 by willprice

hvella, 2016-11-28:
Dear willprice, thank you for this omxWrapper very usefull.
I use it with Raspberry PI under Python application.
I have a question about : how to let the video on screen when the video finsihed to play all the video. ?
I use a screen window size args[--win,'0,50,640,400'] that works fine. But the video automaticcaly close at end of playing. I have tried to use --loop to get the video looping but this does not work.
I would like to get the last frame of video on the window until i use play button again or decide to load antoher video.
Thank you for your support.
Best regards