willprice/python-omxplayer-wrapper #18: Daemon Mode kinda

eN0Rm, 2015-06-11:
Is it possible to make the wrapper pause the video at the end of the file, instead of exiting. I made a hack by adding the --loop option to the omxplayer, so the omxprosess never ends, but it's not optimal.

willprice, 2015-06-11:
Hi eN0Rm,

That seems like a sensible suggestion. I don't have time to implement it in the next week as I'm going on holiday, but once I'm back I'll see whether that is possible.

eN0Rm, 2015-06-12:
No problem, have a nice holiday. If you give me some pointers, I could maybe help out too.

willprice, 2015-06-18:
A few notes:

- A possible implementation could be to query the length of the video and spin up a thread to call pause at that time
- The process might exit for other reasons (i.e. someone skipping forward and playing just before the end)
- Possibly maintain a model of the video in memory and manipulate that as well as the omxplayer process
- Does DBus provide any callback like methods?

eN0Rm, 2015-08-03:
I use a hack. I changed the command that laounches omxplayer.bin to start with the -o parameter. That way the process do not exit.

willprice, 2015-08-03:
Interesting, how does that work? Looking in the docs it seems `-o` is used to specify the output device

eN0Rm, 2015-08-04:
Sorry, I ment the -loop option to omxplayer.bin 
```
    --loop                  Loop file. Ignored if file not seekable
```
https://github.com/popcornmix/omxplayer

eN0Rm, 2015-08-04:
And I used another hack to pause the video after one run. The reason I need this, is that I use a PIR sensor to start the video. So I no people are watching, it wont play.

eN0Rm, 2015-08-22:
Did you give it a go to implement this feature?

willprice, 2015-09-06:
Hi eN0Rm, no I haven't, nor do I think I will since the wrapper's model of the player could be out of sync by a small delta of time meaning that the dbus calls to pause may fail if OMXPlayer has already exited by the time the wrapper thinks the video has finished.

I'm not particularly happy with how the wrapper currently syncs video time as it is hacky and is dependent upon the time it takes OMXPlayer to start. I might hack on it today and see whether I can think of any other way of implementing it.

closed 2015-09-06 by willprice

eN0Rm, 2015-09-07:
Hi, 
With the --loop option, the process do not exit.