willprice/python-omxplayer-wrapper #19: Run One Video and then Loop another

MydKnight135, 2015-06-19:
I have a programming need to play a video based on a keypress, and then if no other keypress is detected after that video finishes, to loop a second video. I'm pretty sure that the functions in your wrapper thus far do not support construction of this type. Is there some enhancement possibily that could enable it?

willprice, 2015-06-19:
I'm doing something quite similar with it actually. Concatenate all your videos into one big file and then skip about the big file thus emulating different video playback. This method avoids the computationally expensive start/stop of OMXPlayer

MydKnight135, 2015-06-19:
That would work from getting between the videos, but how do i get the
effect of looping once i reach the 'end' of one video without getting a
response to play another one?

willprice, 2015-06-19:
Sorry, but I don't understand, can you provide a concrete example of what you want to happen in terms of ... say two videos and the events that cause them to play and the expected behaviour.

MydKnight135, 2015-06-19:
Sure, so lets say we have three 1 minute videos. If I type "1" on the
keyboard, it playes the first video (or in your example, seeks to the start
index of the video on the video glob). if i press 2, it goes to 2's index
and so on. The challenge becomes if video one plays out (reaches the 1
minute mark) I want to, instead of starting video two, play a looping 10
second video as a screensaver until new input is detected. In pseudocode

if key 1:
  play video 1
  wait video1.length
  play screensaver --loop
if key 2:
  playvideo2
  waitvideo2.length
  playscreensaver.loop

and these ifs would be interrupted at any time by a new keypress which
would start that particular if block over.

Make sense?

willprice, 2015-06-19:
Yep, I understand now. 

The way I'd implement that is by spinning up a few threads that sleep for the duration of the video just after you've set the player to execute which would then interact with another thread that controls the screensaver playback.

MydKnight135, 2015-06-19:
I'm trying to understand the interaction points in your suggestion.

Case 1:
  Video thread has executed and is in sleep mode, waiting for its vid to
play. User hits another key

In this case id, what? fire some sort of command to kill that thread so
that after the wait it doesnt start playing the screensaver? Killing
htreasds seems to be something of a sticking point...or would i somehow
"refresh" the video playing thread to play the new video and then reset the
timer accordingly?

case 2:
  screensaver is in looping mode and main program recieves input.

in this case I would somehow need to stop the screensaver thread and start
up the 'play video' thread? right?

There's probably going to be a lot of trickiness to this, but if you can
confirm that my baseline understanding is right and perhaps clarify some
areas that im fuzzy, your suggestion seems to make sense.

Its still surprising to me that this would be as tough as it is...the
blocking behavior of raw_input() has been something of a nightmare for me.

willprice, 2015-06-19:
You're going to want to have one daemon thread that handles the screensaver when a video has finished, and then you'll want to spin up a thread that will:
1. Make the screensaver thread sleep until you wake it up
2. Take control of the player and play the video 
3. Sleep for the same amount of time as the video
4. Wake up the screensaver thread
5. Terminate gracefully.

Make sense?

Threads are pretty icky, processes are nicer, but I haven't investigated them in python.

Off the top of my head that's the best way to do it with the current state of the wrapper. I'm not in a great state of mind at the moment, so it could be that I'm over complicating things... I'll get back to you if I think of a simpler method.

Does that make sense?

MydKnight135, 2015-06-19:
Makes sense. Scares the hell out of me given my understanding of Python,
but it makes sense. I'll start really getting familiar with threads

willprice, 2015-06-19:
Don't be afraid! They're quite simple, but they throw up a bunch of hard to solve problems, race conditions being the most irritating. 

A program that runs multiple threads a, b, and c means that you have three bits of code running concurrently (at the same time, wikipedia for a denser more accurate definition), that's it. The *only* (albeit a hard one) problem is how do you communicate between those bits of code when you need to have x occur in thread a before y in thread b, this is called *synchronisation*, and you can use a handshake to do this, i.e. thread b asks thread a whether it has done x, once thread a has done x, it responds to thread b which can then continue and perform y guaranteeing that x happens before y.

closed 2015-06-19 by willprice