willprice/python-omxplayer-wrapper #20: OMXPlayer object starts in play mode

amaglamatedclyde, 2015-06-30:
when i execute this line in the interpreter:
```player = OMXPlayer('my_file.mp4') ```

the player immediately begins playing the file, instead of starting in pause mode as stated in the docs

clyde

willprice, 2015-06-30:
Hmmm, what raspberry pi are you using and what is the gpu memory split?

amaglamatedclyde, 2015-06-30:
> mode (#20)
> <python-omxplayer-wrapper@noreply.github.com>

hi-it's a first gen model B.mem split set to 128.i have a second gen B i can test.i thought maybe the sleep value in the OMXPLayer init might be too long so it was starting before the self.pause()clyde

willprice, 2015-06-30:
Oh yes, that's almost certainly the issue, it's a horrible hacky way of pausing as there is no true synchronisation. I need to look into DBus to see if I can query the existence of objects and wait until the player has been registered before sending the pause command.

amaglamatedclyde, 2015-06-30:
yes, that makes sense.i did some tests with the new rpi B ver2. python-dbus is included in the noobs image now.when i created the OMXPlayer object in the python interpreter the movie opened in pause mode but also full-screen, so i couldn't enter any more lines in the interpreter.then i tried with ssh into the pi from another  another machine, and running python and entering commands into the interpreter. the movie opened in pause mode, but then i was unable to enter any further commands. maybe i just don't understand how it's meant to operate?thanks!!clyde

amaglamatedclyde, 2015-07-01:
just an update- 

i wrote some code in idle and ran it and everything worked as expected. this is on the RPi 2 model B.
i guess line-by-line use of the interpreter is not the way to work. i can live with that. 

thanks!

willprice, 2015-07-05:
Hi @amaglamatedclyde,

You're correct, line by line use isn't tested, I'm only using this in programs I've written.  

I think I've observed some bugs that might be similar to what you've seen, so I'll push fixes if I can figure out whether it's the client code or this wrapper.

closed 2015-09-06 by willprice