willprice/python-omxplayer-wrapper #23: Can you add a way to enter full screen?

jasonfharris, 2015-07-07:
Hi There,

Could you add a way to enter full screen? I see in player.py can_set_fullscreen(self):

but I don't see the corresponding set_fullscreen(self) method...

Thanks in advance!

willprice, 2015-07-07:
Hi Jason, doesn't OMXPlayer always open in full screen anyway? I don't write understand what you're after. 

Cheers

jasonfharris, 2015-07-08:
Actually it seems to open at some default resolution... I did find via the command line options you can specify the window size: So e.g. from the command line I can call:

omxplayer --win '0 0 1920 1080' --no-osd my_video.avi

(See http://hackersome.com/p/popcornmix/omxplayer for a list of the options some of which are not listed in the command line help e.g. the no onscreen display of the controls which is useful when you are controlling the video programatically.)

Using the above setting I can open the video full screen at 1080p. It would be nice to be able to do that with python-omxplayer-wrapper.

Thanks!
   Jason

willprice, 2015-07-08:
Hi Jason,

You could create a OMXPlayer object using `args` so `omxplayer --win '0 0 1920 1080' --no-osd` would become `OMXPlayer(name, args=['--win', '0 0 1920 1080', '--no-osd'])`

Does this help?

jasonfharris, 2015-07-08:
Ahh... That helps! Thanks. I guess this sort of example should be documented somewhere!
Thanks! Jason

closed 2015-07-08 by jasonfharris

willprice, 2015-07-13:
You're more than welcome to add said documentation in a pull request ;)