willprice/python-omxplayer-wrapper #24: OMXPlayer not running commands

Mohitsharma44, 2015-07-14:
Hello Will,

I was trying to use your wrapper by calling OMXPlayer(<filename>). After executing this command, I can see a small banner flashed at bottom -- Pause 00:00/ 00:23. After this I cannot type anything in the terminal. I have to kill the omxplayer process and my ipython session for the terminal to become responsive again.
Any suggestions?

-Mohit

willprice, 2015-07-15:
This is a known bug, although I'm not sure how to debug it. The module was developed for use in a program so I haven't invested the time in investigating it. I'd welcome help and pull request though ;)

Mohitsharma44, 2015-07-15:
I will see if I can fix it as soon as I get some time.
Thanks

willprice, 2015-09-06:
Is this fixed by passing `args=['--no-keys']` in the constructor? This will prevent OMXPlayer from reading the keyboard input which I suspect might be causing the bug.

phatbank, 2016-03-11:
Hello,

Is it possible to hide the play / pause & time wording on the screen when it plays a video?

willprice, 2016-03-13:
I don't think so, sorry. 

If you can toggle it via a key then you can send the command via `player.action(keys.ACTION)` have a look in the keys file: https://github.com/willprice/python-omxplayer-wrapper/blob/develop/omxplayer/keys.py

willprice, 2016-03-13:
Alternatively there might be some startup flag you can pass to OMXPlayer, which you can inject into the wrapper using the `args` kwarg.

phatbank, 2016-03-13:
Args is the winner. --no-osd

Thanks man!

willprice, 2016-09-12:
Apparently on quit this should be fixed by terminating with `SIGINT` rather than `SIGTERM`. When OMXPlayer is running it takes control of the input devices, I'm not sure whether we can prevent this or not.

willprice, 2016-09-13:
Reading the OMXPlayer docs lists a CLI option: `--no-keys` that disable key input, this should enable you to use the python module in a REPL :)

closed 2016-09-13 by willprice