willprice/python-omxplayer-wrapper #25: Stop video1 and play video2 then continue video1

jonhdeppp, 2015-07-22:
Hi Will,

Hope you can help me with my problem.

video1 = duration 30 secs
video2 = duration 10 secs

I have a looping video1 but in the middle of the play, I stop the video1 and play video2. Once video2 is done, I want to continue video1 on the position it was stopped.

My script::
loop.py
############
player1 = OMXPlayer('video1.mp4')
dur1 = player1.duration()
while 1:
    player.play()
    sleep(dur1)
    player.quit()

stop.py
###########
player2 = OMXPlayer('video2.mp4')
subprocess.call(['dbuscontrol.sh','stop'])
player2.play()

Im having problem on the timer. Is there any other option to replace the time.sleep() in playing looping videos? What happens is when I stopped video1 at 5secs position, I will have a black screen for 15 secs before it plays again and if I stopped video1 at 25secs position, I will experience video1 and video2 playing simultaneously after 5 secs.

Hope you  read this in time.

Thanks,

Joel

willprice, 2015-07-22:
Hi Joel,

I think you're best off joining the two video files into one, and then jumping around that video by using `set_position` this way you avoid having two `OMXPlayer` objects which can cause strange framebuffer artefacts (although someone has mentioned that omxplayer has a --layer flag (or similar) that you can use to layer videos)

jonhdeppp, 2015-07-22:
Hi Will,

Thanks for the quick reply.
If I join the two files, how can I stop video1? Playing video2 is triggered by an event so the timing may differ depending on the trigger.
I tried, the layer option. It kinda solve the few seconds left to be played but still same issue if I have longer time left before I stopped video1.

willprice, 2015-07-22:
Play video1 on loop until the event occurs which you want to play video2, when that happens get the current position of playback and store this, play video2 and when that has finished set the player back to the recorded position. You might find it helpful to model a video as an object where you encapsulate some of the complexity of dealing with multiple videos in one file.

jonhdeppp, 2015-07-22:
One last thing, what is the best way to kill the session of your wrapper? Thanks a lot

willprice, 2015-07-22:
There is a quit method that handles clean up of the process

jonhdeppp, 2015-07-22:
does player.quit() different from "dbuscontrol stop"?

willprice, 2015-07-22:
Yes, check the source code for details. It sends SIGTERM to the process as
well

closed 2015-09-06 by willprice