willprice/python-omxplayer-wrapper #26: Add a "get_filename" function

lbalogh, 2015-08-09:
Hi,

Can you add a "get_filename" option, to get the current playing file ? I would like to do something like :

```
filename = player.get_filename()
```

Thanks !

willprice, 2015-08-09:
Hi Ibalogh,

I'm in two minds about this.
1. Why not just assign the filename to a variable since the player object can't change the file that is being played?
2. Since the player can't change which file it's playing, should it know the filename?

If you can convince me it's a good idea, I'll add it. What's your use case for it?

lbalogh, 2015-08-09:
Hi Will,

You're right, I can just set a variable with the file currently playing, in fact that's what I did ;) But I thought it would be cleaner to get the information from the player object than from an other variable.

My use case is a web server that receive commands from an external app (an iPhone app, actually) to control omxplayer. And I would like to query the webserver to know if something is playing, and if yes, respond with the filename currently playing.

It's here : https://github.com/lbalogh/OMXPlayer-Remote/blob/master/OMXPlayerServer.py, under "elif command == 'is_playing':" (this is a work in progress and needs cleanup) if you want to take a look

willprice, 2015-08-12:
I think you're probably right, I'll try and get round to it in the next couple of weeks.

lbalogh, 2015-08-13:
Great ! Thanks ;)

willprice, 2015-08-13:
If you're feeling adventurous you could do it yourself and submit a pull
request

willprice, 2015-09-06:
Merged in #27

closed 2015-09-06 by willprice