willprice/python-omxplayer-wrapper #28: hide the status and time information at movie start

electricgraphy, 2015-08-18:
Hi Will

thank you for your wonderful source.
is there any method to hide the status and time information of movie when it starts?

Jin

willprice, 2015-08-18:
You can pass command line arguments in the `OMXPlayer` constructor, perhaps `--no-osd` is what you're looking for?

electricgraphy, 2015-08-18:
thank you for the quick reply.

sorry I'm just newbie on python and linux. ;)  
I tried to make some video-installation between 2 raspberry pi that must communicate each other.
so the platform must be python.

the part of omxplayer in my code is so simple.
just like this:
player = OMXPlayer('path/to/file.mp4')
player.play()
sleep(player.duration())
player.quit()

I wonder if you could explain in detail how I control or use the arguments in the OMXPlayer.
is it possible with this code?  

thank you very much
Jin

willprice, 2015-08-18:
initialise with `OMXPlayer('path.mp4', args=['--no-osd'])`

electricgraphy, 2015-08-18:
Thank you so much! It work. :)

closed 2015-08-18 by electricgraphy