willprice/python-omxplayer-wrapper #29: Stop information overlay from showing.

NonaSuomy, 2015-08-24:
How do you stop player from overlaying any information on the screen (Playing, Time, etc) just want video nothing else

willprice, 2015-08-24:
Please look through previously closed issues as this has already come up
and been solved

closed 2015-08-24 by willprice

NonaSuomy, 2015-08-25:
I found this initialise with OMXPlayer('path.mp4', args=['--no-osd']) how do I add multiples?

initialise with OMXPlayer('path.mp4', args=['--no-osd --no-keys -b'])   ?

This gave me an error saying: unrecognised options '--no-osd --no-keys -b'

NonaSuomy, 2015-08-25:
I tried OMXPlayer('path.mp4', args=['--no-osd'], args=['--no-keys'], args=['-b'])
No luck, SyntaxError: keyword argument repeated.

Just to note I've tested the options directly with the player and they work fine ie: omxplayer --no-osd --no-keys -b path.mp4 etc

NonaSuomy, 2015-08-25:
This may have worked for future people with this issue: OMXPlayer('path.mp4', args=['--no-osd', '--no-keys', '-b'])