willprice/python-omxplayer-wrapper #32: Enough dbus-daemon --forks to fill a drawer

NonaSuomy, 2015-08-27:
root      2668  0.0  0.3   3184  1780 ?        Ss   Aug26   0:00 dbus-daemon --fork --print-address 5 --p
root      2719  0.0  0.3   3184  1780 ?        Ss   Aug26   0:00 dbus-daemon --fork --print-address 5 --p
root      2854  0.0  0.3   3184  1748 ?        Ss   00:49   0:00 dbus-daemon --fork --print-address 5 --p
root      2874  0.0  0.4   3184  1856 ?        Ss   00:50   0:00 dbus-daemon --fork --print-address 5 --p
root      2946  0.0  0.4   3184  1852 ?        Ss   01:13   0:00 dbus-daemon --fork --print-address 5 --p
root      2987  0.0  0.4   3184  1852 ?        Ss   01:15   0:00 dbus-daemon --fork --print-address 5 --p
root      3049  0.0  0.4   3184  1852 ?        Ss   01:17   0:00 dbus-daemon --fork --print-address 5 --p
root      3069  0.0  0.4   3184  1848 ?        Ss   01:18   0:00 dbus-daemon --fork --print-address 5 --p
root      3120  0.0  0.3   3184  1776 ?        Ss   01:19   0:00 dbus-daemon --fork --print-address 5 --p
root      3140  0.0  0.3   3184  1716 ?        Ss   01:19   0:00 dbus-daemon --fork --print-address 5 --p
root      3160  0.0  0.3   3184  1724 ?        Ss   01:19   0:00 dbus-daemon --fork --print-address 5 --p
root      3180  0.0  0.4   3184  1852 ?        Ss   01:19   0:00 dbus-daemon --fork --print-address 5 --p
root      3200  0.0  0.4   3184  1848 ?        Ss   01:20   0:00 dbus-daemon --fork --print-address 5 --p
root      3253  0.0  0.4   3184  1852 ?        Ss   01:31   0:00 dbus-daemon --fork --print-address 5 --p
root      3273  0.0  0.4   3184  1852 ?        Ss   01:31   0:00 dbus-daemon --fork --print-address 5 --p
root      3293  0.0  0.3   3184  1748 ?        Ss   01:31   0:00 dbus-daemon --fork --print-address 5 --p
root      3314  0.0  0.4   3184  1848 ?        Ss   01:40   0:00 dbus-daemon --fork --print-address 5 --p
root      3344  0.0  0.0      0     0 ?        S    01:47   0:00 [kworker/0:0]
root      3345  0.0  0.4   3260  2016 tty1     Ss   01:47   0:00 /bin/login -f      tty1
root      3360  0.9  0.6   4788  2752 tty1     S+   01:58   0:00 sudo python ./6wohspg.p
root      3361 68.4  1.8  27152  8072 tty1     Rl+  01:58   0:30 python ./6wohspg.py
root      3384  0.1  0.3   3184  1720 ?        Ss   01:58   0:00 dbus-daemon --fork --print-address 5 --p
root      3404  0.2  0.3   3184  1776 ?        Ss   01:58   0:00 dbus-daemon --fork --print-address 5 --p
root      3424  0.3  0.3   3184  1772 ?        Ss   01:59   0:00 dbus-daemon --fork --print-address 5 --p
root      3444  0.4  0.4   3184  1784 ?        Ss   01:59   0:00 dbus-daemon --fork --print-address 5 --p
root      3457  0.7  0.6   4316  2768 ?        Ss   01:59   0:00 /bin/bash /usr/bin/omxplayer --no-osd /h
root      3464  0.6  0.3   3184  1748 ?        Ss   01:59   0:00 dbus-daemon --fork --print-address 5 --p
root      3467 31.7  6.1 122252 27464 ?        Sl   01:59   0:02 /usr/bin/omxplayer.bin --no-osd 

Every Time I start and quit the player seems like it creates more. Is there a way to clean this up after a quit?

```python
from time import sleep
from omxplayer import OMXPlayer
videos= ['/home/wohs/src/i2c/big_buck_bunny_480p_H264_AAC_25fps_1800K.MP4']
import RPi.GPIO as GPIO
pinSwi00=23
GPIO.setmode(GPIO.BCM)
GPIO.setup(pinSwi00, GPIO.IN, pull_up_down=GPIO.PUD_UP)
playing = False
index = 0
omx=None
def my_callback(channel):
	print 'test1'
	global playing, index, omx
	try:
		if GPIO.input(pinSwi00):
			index = 0
			if playing:
				sleep(0.5)
				omx.quit()
				playing = False
				print "quitting"
				sleep(0.5)
				omx=OMXPlayer(videos[index], args=['--no-osd'])#,'-b','--layer 2','--loop'])
				omx.play()
				playing = True
				sleep(0.5)
			else:
				sleep(0.5)
				omx=OMXPlayer(videos[index], args=['--no-osd'])#,'-b','--layer 2','--loop'])
				omx.play()
				playing = True
				sleep(0.5)
			print "playing..."
	except Exception as inst:
		print type(inst)
		print inst.args      # arguments stored in .args
		print inst           # __str__ allows args to be printed directly

GPIO.add_event_detect(pinSwi00, GPIO.RISING, callback=my_callback, bouncetime=300)

def main():
	global omx, playing
	try:
        	while True:
	                continue
	except KeyboardInterrupt:
		GPIO.cleanup()
		if playing:
			omx.quit()
if __name__ == "__main__":
        main()
```

willprice, 2015-09-06:
Crikey.

That doesn't look good. Are you sure the wrapper is forking these calls? I'll have a look into it shortly.

RaulGuichard, 2015-09-08:
Exactly the same here, made a script that loops between two videos, each time a video finishes playing it calls the quit() method and loads that video again while the other is playing and so on, and so on.

And I can see how it keeps creating new dbus-daemon processes:

"dbus-daemon --fork --print-address 5 --print-pid 6 --session"

I count them with this line and they increase with time, till it takes all the memory and hangs

ps ux | grep dbus-daemon | grep -v grep | wc -l

Not an expert on dbus but it seems like it keeps spawning new dbus processes and not killing nor reusing them.

willprice, 2015-09-08:
The `DBusConnection` class creates a new bus connection per instantiation but doesn't clean up afterwards, I'll look into added a deconstructor to clean up. Currently I can't do this as I'm waiting on some parts for my RPi, but once I get them I'll look into it.

Thanks for the comment.

willprice, 2016-09-12:
I need to investigate this DBus memory leak, I think it's due to creating a new DBus connection each time bus finder runs: `dbus.bus.BusConnection`. I need to test with `dbus.SessionBus()` to see whether this causes new connections or not.

I have reverted the contributed code as it killed or DBus daemons which messes with the tests and could make people's systems unstable.

willprice, 2016-10-08:
I've tracked down the root cause of this.

When `omxplayer` is launched, if `/tmp/omxplayer.<USER>.pid` doesn't exist, then `dbus-daemon` will be launched. When `omxplayer` exists, it doesn't clean up the `dbus-daemon` process as it is expected that `omxplayer` may be run again, and it can reuse the existing session. On subsequent launches, `omxplayer` looks for `/tmp/omxplayer.<USER>.pid`and finds that the file exists and so reuses the existing `dbus-daemon` process.

We currently clean the `/tmp/omxplayer*` files on quitting, but don't kill the `dbus-daemon` process, hence the proliferation of `dbus-daemon` processes.

Rather than killing the spawned `dbus-daemon` (fighting against `omxplayer`) we might as well leave it be, and leave the files intact rather than cleaning them up.

commit b5b94f08 (2016-02-18): Merge pull request #36 from RaulGuichard/develop

Fixes #32, dbus processes memory leak
commit 7d6c421b (2016-10-08): Bugfix: Prevent `dbus-daemon` spawning for new instances of `OMXPlayer`

When `omxplayer` launches, it looks for two files in `/tmp`,
* `omxplayer.<USER>.pid`, holds the PID of the `dbus-daemon` launched by
  `omxplayer`.
* `omxplayer.<USER>`, holds the address of the `omxplayer` on
  `dbus`

When these files exist, `omxplayer` reuses the existing bus, and doesn't
launch another, however we were previously cleaning up these files
causing the instance of `omxplayer` launched to spawn another
`dbus-daemon`. By preserving these files we avoid spawning another
`dbus-daemon` for each instantiation  [... cut]
PR #55 (2016-10-08): Bugfix/dbus uncontrollable spawning
## Description
* Fix #37 (same as #40)
* Fix #32

* **Status**: *READY*
commit 81e43736 (2016-10-08): Bugfix: Prevent `dbus-daemon` spawning for new instances of `OMXPlayer`

When `omxplayer` launches, it looks for two files in `/tmp`,
* `omxplayer.<USER>.pid`, holds the PID of the `dbus-daemon` launched by
  `omxplayer`.
* `omxplayer.<USER>`, holds the address of the `omxplayer` on
  `dbus`

When these files exist, `omxplayer` reuses the existing bus, and doesn't
launch another, however we were previously cleaning up these files
causing the instance of `omxplayer` launched to spawn another
`dbus-daemon`. By preserving these files we avoid spawning another
`dbus-daemon` for each instantiation  [... cut]
closed 2016-10-08 by willprice