willprice/python-omxplayer-wrapper #33: Does not install correctly on Rasperry Pi

WilliWichtel, 2015-09-13:
Hi,

Itried to install this on my RasPi, the setup completed with no errors:

me@raspberrypi ~/temp/omxplayer-wrapper-0.0.2 $ sudo python3 setup.py install
running install
Checking .pth file support in /usr/local/lib/python3.2/dist-packages/
/usr/bin/python3 -E -c pass
TEST PASSED: /usr/local/lib/python3.2/dist-packages/ appears to support .pth files
running bdist_egg
running egg_info
writing requirements to omxplayer_wrapper.egg-info/requires.txt
writing omxplayer_wrapper.egg-info/PKG-INFO
writing top-level names to omxplayer_wrapper.egg-info/top_level.txt
writing dependency_links to omxplayer_wrapper.egg-info/dependency_links.txt
reading manifest file 'omxplayer_wrapper.egg-info/SOURCES.txt'
writing manifest file 'omxplayer_wrapper.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv6l/egg
running install_lib
running build_py
creating build/bdist.linux-armv6l/egg
creating build/bdist.linux-armv6l/egg/omxplayer
copying build/lib/omxplayer/bus_finder.py -> build/bdist.linux-armv6l/egg/omxplayer
copying build/lib/omxplayer/player.py -> build/bdist.linux-armv6l/egg/omxplayer
copying build/lib/omxplayer/keys.py -> build/bdist.linux-armv6l/egg/omxplayer
copying build/lib/omxplayer/__init__.py -> build/bdist.linux-armv6l/egg/omxplayer
copying build/lib/omxplayer/dbus_connection.py -> build/bdist.linux-armv6l/egg/omxplayer
byte-compiling build/bdist.linux-armv6l/egg/omxplayer/bus_finder.py to bus_finder.cpython-32.pyc
byte-compiling build/bdist.linux-armv6l/egg/omxplayer/player.py to player.cpython-32.pyc
byte-compiling build/bdist.linux-armv6l/egg/omxplayer/keys.py to keys.cpython-32.pyc
byte-compiling build/bdist.linux-armv6l/egg/omxplayer/__init__.py to __init__.cpython-32.pyc
byte-compiling build/bdist.linux-armv6l/egg/omxplayer/dbus_connection.py to dbus_connection.cpython-32.pyc
creating build/bdist.linux-armv6l/egg/EGG-INFO
copying omxplayer_wrapper.egg-info/PKG-INFO -> build/bdist.linux-armv6l/egg/EGG-INFO
copying omxplayer_wrapper.egg-info/SOURCES.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying omxplayer_wrapper.egg-info/dependency_links.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying omxplayer_wrapper.egg-info/requires.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying omxplayer_wrapper.egg-info/top_level.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating 'dist/omxplayer_wrapper-0.0.2-py3.2.egg' and adding 'build/bdist.linux-armv6l/egg' to it
removing 'build/bdist.linux-armv6l/egg' (and everything under it)
Processing omxplayer_wrapper-0.0.2-py3.2.egg
Removing /usr/local/lib/python3.2/dist-packages/omxplayer_wrapper-0.0.2-py3.2.egg
Copying omxplayer_wrapper-0.0.2-py3.2.egg to /usr/local/lib/python3.2/dist-packages
omxplayer-wrapper 0.0.2 is already the active version in easy-install.pth

Installed /usr/local/lib/python3.2/dist-packages/omxplayer_wrapper-0.0.2-py3.2.egg
Processing dependencies for omxplayer-wrapper==0.0.2
Searching for decorator==4.0.2
Best match: decorator 4.0.2
Processing decorator-4.0.2-py3.2.egg
decorator 4.0.2 is already the active version in easy-install.pth

Using /usr/local/lib/python3.2/dist-packages/decorator-4.0.2-py3.2.egg
Finished processing dependencies for omxplayer-wrapper==0.0.2
me@raspberrypi ~/temp/omxplayer-wrapper-0.0.2 $

But when I invoke the import function then there ist no such package:

Python 3.2.3 (default, Mar  1 2013, 11:53:50) 
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from omxplayer import OMXPlayer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.2/dist-packages/omxplayer_wrapper-0.0.2-py3.2.egg/omxplayer/__init__.py", line 1, in <module>

>>> 

I found found the following egg:
/usr/local/lib/python3.2/dist-packages/omxplayer_wrapper-0.0.2-py3.2.egg
but no corresponding files or folders like other packages (wheel, pexpect,virtualenv,..) do.
So I tried to unzip that egg but it still does not work.

I'm new to python so I don't have a clue what to do. Could you give me an hint please?

Cheers, Falk

willprice, 2015-09-13:
> File "", line 1, in

Hi Falk, currently this is only a python 2 package. I haven't gotten around
to adding python three support yet

WilliWichtel, 2015-09-13:
much better now ;-)

closed 2015-09-13 by WilliWichtel