willprice/python-omxplayer-wrapper #34: how to I add args?

electricgraphy, 2015-09-20:
Hi
sorry for a stupid ask but I really don't know about python..

>> player = OMXPlayer('/home/pi/03.mp4', args=['--no-osd'])

I want to add another args, for example '-o local'.
I tried like this but it does not work.

>> player = OMXPlayer('/home/pi/03.mp4', args=['--no-osd','-o local'])

can you help me?
thank you very much.

willprice, 2015-09-20:
Hi,

You have to obey the shell word splitting rules, you can use the shlex
library to split a string of args into an array of words the same as how
the shell would split them.

Your mistake is to include local with -o, you want [ '-o', 'local' ]

electricgraphy, 2015-09-20:
Thanks for your quick reply.
I tried again as your replay, like this:
>> player = OMXPlayer('/home/pi/03.mp4', args=['-o','local'])
but it does not work too with some error message.
would you check this one more time?
thank you.

willprice, 2015-09-20:
Can you provide the error message please.

electricgraphy, 2015-09-21:
Hi when I tried one more time with same code today, It works!
Thank you very much :D

closed 2015-10-23 by willprice