willprice/python-omxplayer-wrapper #35: Checking if file exists prevents playing streams

ghtyrant, 2015-10-21:
In https://github.com/willprice/python-omxplayer-wrapper/blob/develop/omxplayer/player.py#L100-L101 you check if the file to be played exists in the filesystem. OMXPlayer is able to play streams (HLS, RTSP, ...) and this check fails for stream URLs (e.g. http://example.com/livestream)

willprice, 2015-10-22:
Thanks for the heads up @ghtyrant, I didn't know OMXPlayer had that functionality! What do you think I should do? Check to see if the input is file like if so check the file and exists and otherwise don't perform the check? Or should I just remove the check completely and try and tidy up the mess that is made after OMXPlayer fails to launch?

ghtyrant, 2015-10-22:
I would simply remove that check and let OMXPlayer handle it - though, I don't know much about the player's internals :)

tiangolo, 2016-07-13:
@ghtyrant could you check my PR while @willprice sees it? 

To test locally I think you would just have to:

```
git clone https://github.com/tiangolo/python-omxplayer-wrapper.git
cd python-omxplayer-wrapper
git checkout feature/support-url
sudo pip install --upgrade ./
```

And then use as normally, feeding it a RTSP (for example streaming a local file with VLC):

```python
from omxplayer import OMXPlayer

player = OMXPlayer('rtsp://192.168.1.15:/8554/')
```

willprice, 2016-07-13:
Thanks @tiangolo for the pull request. Once you get someone else to validate that it works I'll merge it in. Your help is greatly appreciated :+1:

willprice, 2016-10-08:
(See pull request for rejection reason)

closed 2016-10-08 by willprice