willprice/python-omxplayer-wrapper #37: Frozen console

TuomLarsen, 2015-10-29:
Hello!

When I try to run python smoke-tests/test2.py, it plays the videos but when it exits I cannot type in console anymore. More precisely, I can type but nothing shows up but it's there somewhere because when I (blindly) hit up-arrow key and enter, it stars playing again. It's just the console wont display anything I type after I run python-omxplayer-wrapper.

I'm on RPi 2, Raspian Jessie 2015-09-24.

willprice, 2015-10-31:
Hi @TuomLarsen, This is a known issue, I'm not quite sure what causes it though. Initially I thought it might be OMXPlayer reading key presses and when it quits it doesn't return focus back to the shell, but as you've pointed out, the shell is receiving characters, they just aren't appearing.

As a crap work around I tend to run REPLs in tmux so that I can kill the pane and create a new one quickly--sorry I don't have a better response.

Aminpro, 2016-01-02:
Please refer to Pull Request #40 
That fixed the problem.

willprice, 2016-01-03:
Fixed in #40

willprice, 2016-10-08:
I can still observe this issue in the `develop` branch, redirecting stdin to `devnull` leaves the console usable. I think this is a more robust fix than killing with `SIGINT` as it allows the console to be used during interactive sessions.

commit 04f8d46e (2016-10-08): Bugfix: Terminal remains usable once `omxplayer` has been spawned

Previously `omxplayer` took over `stdin` resulting in the following two
issues:
* When running in a REPL the user would no longer be able to interact
  with the REPL until `omxplayer` quit.
* Once `omxplayer` had quit, the terminal would no longer show
  characters despite registering them (so you could type commands and
  execute them, but they wouldn't appear on the screen)

Both problems were resolved by redirecting `stdin` of the process to
`/dev/null`.

Fixes #37
PR #55 (2016-10-08): Bugfix/dbus uncontrollable spawning
## Description
* Fix #37 (same as #40)
* Fix #32

* **Status**: *READY*
commit 2c2fba9d (2016-10-08): Bugfix: Terminal remains usable once `omxplayer` has been spawned

Previously `omxplayer` took over `stdin` resulting in the following two
issues:
* When running in a REPL the user would no longer be able to interact
  with the REPL until `omxplayer` quit.
* Once `omxplayer` had quit, the terminal would no longer show
  characters despite registering them (so you could type commands and
  execute them, but they wouldn't appear on the screen)

Both problems were resolved by redirecting `stdin` of the process to
`/dev/null`.

Fixes #37
closed 2016-10-08 by willprice