willprice/python-omxplayer-wrapper #39: Two videos script fail with dbus.exceptions.DBusException

argrento, 2015-12-16:
```python
from omxplayer import OMXPlayer
import time

launchParameters = ['--blank', '-o', 'hdmi', '--no-osd',
                    '--win', '0,0,1279,719']
player1 = OMXPlayer('/home/pi/Videos/video1.mp4', args=launchParameters)
video1Duration = player1.duration()

player2 = OMXPlayer('/home/pi/Videos/video2.mp4', args=launchParameters)
video2Duration = player2.duration()

print("first...")
player1.play()
time.sleep(10)
player1.quit()
print("second...")
player2.play()
time.sleep(10)
player2.quit()
print("done!")
```

When I run this script, it fails with an error.
```python
Traceback (most recent call last):
  File "test.py", line 9, in <module>
    player2 = OMXPlayer('/home/pi/Videos/video2.mp4', args=launchParameters)
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 73, in __init__
  File "<decorator-gen-20>", line 2, in pause
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 134, in wrapped
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 309, in pause
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)

```

That was really strange, I suggested that videos are broken. So I run this code with another videos and no error was detected. 
Btw: omxplayer can play all this files.

JonnyAlpha, 2015-12-28:
agrento;

I am in no way an expert but I used OMXPlayer last year and have just started my project with it again.
In the absence of any other replies (its the holidays), for de-bugging have you tried your code without using the launchParameters?

Also what version of Python are you using, I had some issues with DBUS / different versions of Python until I ran a patch to setup DBUS to work with Python 3.

These are just guesses of course :-)

willprice, 2016-02-18:
Two player objects are not able to exist simultaneously -- this is unsupported behaviour.

closed 2016-02-18 by willprice