willprice/python-omxplayer-wrapper #42: Request: changing playback speed while playing video

deanthinker, 2016-01-27:
Hi, not sure if I come to the right place for feature request. Is it possible to change the playback rate while the video is playing? Thanks.

willprice, 2016-01-29:
You could have a go using the `action` method with the keybindings `1` or `2`, the [docs](https://github.com/popcornmix/omxplayer) seem to suggest that those control speed, but omxplayer doesn't expose any dbus interface to control speed.

willprice, 2016-02-18:
Since I've had no response, I'm assuming this works, please comment & reopen if this is not the case

closed 2016-02-18 by willprice

markkorput, 2016-07-08:
Verified working!

malban, 2018-03-21:
I realize this is old, but I am trying to do something similar and found an alternate method that allows for exact control over the playback speed instead of just the fixed increments:

```
player = OMXPlayer(VIDEO_PATH)
player._player_interface_property('Rate', dbus.Double(0.2))
```

cooleyandy, 2020-01-21:
@malban I tried your code but it doesn't seem to affect the playrate. Are you using the omxplayer that is included with raspbian, or another build?