willprice/python-omxplayer-wrapper #45: Stream over HTTP

jappe999, 2016-05-25:
Hi,

I tried to use this wrapper and I must say it's awesome.
There is only one thing: for some reason I can't stream form another server. When I try the exact same with omxplayer from the command line, it does play.

Do you have any clue about what it could be?

jappe999, 2016-05-25:
I read about it in another issue. So I'm fine. Thank you :)

jappe999, 2016-05-25:
Aaaand I'm back.

So I have messed a little bit in the source code, because I struggled over the fact, that I can't stream over HTTP via the omxplayer-wrapper.

If I comment out the lines where you check if the path exists on your raspberry, there rolls out a new error with the announcement of: "'DBus cannot connect to the OMXPlayer process".

I checked half the source code, but I can't seem to find any clue.

In another issue is asked the same thing, but if I do that, it still doesn't work... :(

willprice, 2016-05-25:
Hi,

Thanks for taking the effort to look for existing issues first, what command can you run manually that works? Can you show me how you're instantiating the `OMXPlayer` object and provide a full stack trace.

jappe999, 2016-05-28:
Hi,

I regularly use the following command in my terminal: 
```
omxplayer -o local "http://example.com/file.mp3" '.
```

I instantiate the OMXPlayer object as followed:
```python
def play();
    player = OMXPlayer('"%s"' % track_location)  # track_location is a http-link to a soundcloudstream.
    player.play()
```

With the check if the file is on the computer disabled, I get this message:
```
DBus cannot connect to the OMXPlayer process
```

If I get the original Errormessage it says:
```
Could not get proxy object
```

jappe999, 2016-06-17:
Hey @willprice,

I finally got it working. The wrapper can be used over HTTP, but you have to open the url with urllib.urlretrieve:

```python
import urllib
from omxplayer import OMXPlayer

file_name = YOUR_FILE_URL

try:
    	urllib.urlretrieve(file_name, 'file.mp3')
    	player = OMXPlayer('file.mp3', ['-o', 'local'])
    	player.play()
except Exception as e:
    	print e
```

closed 2016-06-17 by jappe999

willprice, 2016-06-17:
Hi @jappe999,

Sorry I couldn't be of more help, glad you managed to find a solution, and thanks for taking the time to document it. I'll add it as an example to the README :+1: