willprice/python-omxplayer-wrapper #50: Be able to load new video within the same OMXPlayer instance

markkorput, 2016-09-05:
This would probably still involve killing any existing omxplayer process and spawning a new process, but would make it easier for the caller to;

* run multiple files with the same args without having to manage the args locally. So you could 

do
```python
player = OMXPlayer('video1.mp4', args=['--no-osd', '--adev', 'local', '-b'])
```
and then 
```python
player.load('video2.mp4')
```
and the second video would be played with the same options ('--no-osd --adev local -b').

* create extended functionality in an external class that can simply reference the same OMXPayer instance without having to deal with managing multiple instances (see #49 for an example).

willprice, 2016-09-12:
This would be really nice.

I don't really have the time, nor inclination to continue working on this project, but if you're interested in helping out I'd love to open the project up. I can help with code review and build automation.

Currently the build got broken after merging in a slightly over-zealous pull request that seems to crash the test runner or python instance.

markkorput, 2016-09-12:
I understand, I intend to make a pull request for every "feature request". Note that pull request #52 implements this and it seems to work fine, but I haven't extensively tested it.

willprice, 2016-09-12:
Thanks Mark, I really appreciate your contribution! This project hasn't seen much love for a year or so!

aresstokrat, 2016-09-29:
this is a great idea to put this into project.
If you can do this starting another file without spawning another process woud be even cooler.

markkorput, 2016-10-01:
@aresstokrat that would be ideal,  yea,  unfortunately omxplayer doesn't currently seem to expose an (dvus) interface to load files dynamically so I don't think this is possible. Unless you have suggestions? 

Note that otherwise the load functionality is working and implemented in pull request #52 @willprice any reason what that hasn't been merged in?

willprice, 2016-10-08:
Completed in #57

closed 2016-10-08 by willprice

manojjenago, 2018-02-02:
Hi,
I want to use youtube-dl with OMXPlayer ... Is that possible??
var url = "https://www.youtube.com/watch?v=TN0cxBYh8t8";

                child = spawn('omxplayer', [{$(youtube-dl -g url} , config.position, config.width, config.height], {
                                cwd: 'modules/MMM-Screencast'
}

I know this one alone in shell works 
omxplayer -o hdmi $(youtube-dl -g 'https://www.youtube.com/watch?v=frH9HaQTFL8')

how can I use with spawn -- Please guide..
Thanks
MK

willprice, 2018-02-02:
Hi @manojjenago you'll want to download to a temporary file descriptor and then use this as the file you call from omxplayer. This is an issue tracker for problems with omxplayer-wrapper and not general questions on how to build applications; you'd be better off on the RPi forums seeking this sort of advice.

rioardhityo, 2019-02-09:
> Hi,
> I want to use youtube-dl with OMXPlayer ... Is that possible??
> var url = "https://www.youtube.com/watch?v=TN0cxBYh8t8";
>
> ```
>             child = spawn('omxplayer', [{$(youtube-dl -g url} , [...]

Hllow i was doin so back then, u could, but i didnt used youtube-dl, i was used pytube, so basically paytube is for management youtube stuff inside python code.