willprice/python-omxplayer-wrapper #59: Display Live-Movie from gphoto2

tschiffler, 2016-10-11:
I am just trying your wrapper in my test environment for a few days but i am not able to find a solution for my problem. 

my setup
- raspberry 3
- connected eos digital
- installed gphoto2, omxplayer, omxplayer-wrapper

if I enter on console the following command, the omxplayer is shown with the livemovie of my gphoto2 (file is already written by 'gphoto2 --capture-movie --stdout > /home/abc/fifo.mjpg')
Command: 'omxplayer /home/abc/fifo.mjpg --live'

now I tried to display it in the wrapper with
player = OMXPlayer("/home/abc/fifo.mjpg")
player.play()

After executing i get the FileNotFoundError:
omxplayer.player.FileNotFoundError: Could not find: /home/abc/fifo.mjpg

The error is thrown at player.py in line 106

willprice, 2016-10-11:
Have you tried passing in `--live` as one of the arguments:
`OMXPlayer("/home/abc/fifo.mjpg", args=['--live'])`

tschiffler, 2016-10-12:
> thread [2].
> Links:

Hey,

yes, i already tried this, too
I´ll try again later to be sure and give you a validated answer

Greetings Thomas

willprice, 2016-10-23:
Have you managed to reproduce this @tschiffler?

bknobloch, 2016-11-16:
I believe the reason is that fifp.mjpg is a named pipe resulting in python's os.path.isfile() returning false in player.py line 107.

Maybe adding a check for a pipe might improve:
stat.S_ISFIFO(os.stat(source).st_mode)

willprice, 2017-09-26:
Should be fixed in latest master release.

closed 2017-09-26 by willprice