willprice/python-omxplayer-wrapper #65: "randomly" occuring error when pausing or stopping video

Akwariom, 2016-12-16:
First of all thanks much Will for this great and very easy to use library. 

I am experiencing an apparently randomly happening issue when pausing or stopping a playing video. The command just does nothing and I get the following error message  :

dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.299 was not provided by any .service files

(or any other float instead of 1.299). 

I can't seem to understand when it is happening. most of the time pause() and stop() work like a charm, sometimes they do not. When they do not, most of the time reopening the video with load() solves the issue but sometimes I litteraly have to reopen it 10 times before being able to pause or stop it again. 

Since I am not a great programmer, I need help :-) !
Thanks

jeffbl, 2016-12-19:
I am seeing the same issue, but only if I'm messing around with loading different files. If I have two OXMPlayer instances and I pause one and start another, then quit the first one, the error triggers on pause() almost every time. Probably something I'm doing, but wanted to say I'm seeing this as well. Am working around by using a simpler method to accomplish my task...

willprice, 2016-12-20:
Sorry that you're having issues. I've personally concatenated videos together into a single file, then written a layer over omxplayer to skip around in the video to switch between videos (this worked pretty well).

I imagine this might be caused by omxplayer trampling over the dbus information it stores in /tmp/omxplayer.something (can't remember off the top of my head). Alternatively it could well be a problem in the bus finder that reads information from this file (might be getting out of sync)

jeffbl, 2016-12-21:
Hi Will - thanks for the quick response. That is the approach I am planning to use as well. Do you just use the set_position to do so? Have you ever tried anything with chaptering, and if so, does it work any better / more reliably?

Akwariom, 2016-12-22:
Hi Will and Jeff. Thanks much for your replies. 
Unfortunately I have too many videos and I need to be able to change some of them quickly, so concatenation won't do it for me. It seems that whenever stop() and pause() don't work, set_position() doesn't either. load() is 100% reliable though, so I created a black video that I recall as a workaround for stop().
Other thing, from what I get, it's not possible to change the --loop state of omxplayer without closing and reopening it. Do you see a workaround for that ? Sometimes I'd like to be able to change the loop state while playing a video (don't ask me why ;-)) 
Thanks again !

willprice, 2017-09-26:
I'm pretty sure this is due to omxplayer trashing the `/tmp/omxplayer.*` files which are a massive PITA. See #107 for more. tl;dr use `dbus_name` to avoid issues.

closed 2017-09-26 by willprice