willprice/python-omxplayer-wrapper #66: OMXPlayer Wrapper for Python3?

embgit, 2016-12-19:
Hey Will, any timeline for migrating your wrapper to Python 3 for Rpi? Or possibly, is there a 
current workaround?  I'm stuck with my app working in Python 3 for various reasons. I'm of course getting the error message using your code example: ImportError: No module named 'omxplayer'

I'm really excited about getting away from using subprocess to call Omxplayer and being able to use your package internally just like Picamera. I was looking at the Popcornmix DBUS CONTROL Interfaces and cannot find an equivalent of your set_video_pos() - this will make my app really easy to program different screen sizes include the 7 inch Rpi touch screen.

Thanks in advance

jeffbl, 2016-12-19:
The only problem I had using it with python3 was that I needed to change the url parsing in player.py to be:

     import urllib.parse

and in def _setup_omxplayer_process change the second line to

     source_url = urllib.parse.urlsplit(source)

Now it seems to be working fine on a Pi3 running Ubuntu Mate 16.04

But yes, would be nice if it worked out of the box. Thanks for the great video solution!

embgit, 2016-12-20:
Jeff - I got your solution to work.  I also had to install python3 dbus.

        sudo apt-get install python3-dbus

FYI, I'm running my Pi3 on plain vanilla Rpi OS, so you being on Ubuntu made no
difference in making this work.

You guys have a great package, so now I can add a lot of bells and whistles to the play section of my Kivy application with ease.  I'm using the NOIR camera to record a scintillator screen that is attached my home made interferometer.  I can explain some problems I'm having with low light levels if you want to also solve that problem (camera cannot use IR LED's or any other illuminator to read  a phosphor scintillator screen).

Thanks again, Mark

willprice, 2016-12-20:
Hi @jeffbl @embgit, sorry for the lack of communication, I've been quite busy (final year at uni!).

Glad you managed to get things working, I'd really like to make the package support python 3 as well, glad @jeffbl found a way to do this, if you find  you have any other problems related to python 3 support it would be most helpful if you could log them here. I might take a look at this at the end of January after exams and set up tox to test py3 support.

embgit, 2016-12-23:
No problem.  I got another problem that I fixed myself with python 3 no longer supporting the long function.  I was using your duration() and got the following:

vidtime = player.duration()
player.play()
sleep(vidtime)
 return long(self._get_properties_interface().Duration())
NameError: name 'long' is not defined

embgit, 2016-12-23:
No problem.  I got another problem that I fixed myself with python 3 no longer supporting the long function.  I was using your duration() and got the following:

   vidtime = player.duration()
   player.play()
   sleep(vidtime)

   return long(self._get_properties_interface().Duration())
   NameError: name 'long' is not defined

I of course replayed the long() with and int()

Mark

embgit, 2016-12-23:
No problem. I got another problem that I fixed myself with python 3 no longer supporting the long function. I was using your duration() and got the following:

vidtime = player.duration()
player.play()
sleep(vidtime)

return long(self._get_properties_interface().Duration())
NameError: name 'long' is not defined

I of course replaced the long() with and int()

Mark

embgit, 2016-12-23:
No problem. I got another problem that I fixed myself with python 3 no longer supporting the long function. I was using your duration() and got the following:

vidtime = player.duration()
player.play()
sleep(vidtime)

return long(self._get_properties_interface().Duration())
NameError: name 'long' is not defined

I of course replaced the long() with a int()

Mark

closed 2016-12-27 by embgit