willprice/python-omxplayer-wrapper #67: SystemError: DBus cannot connect to the OMXPlayer process

nsuva, 2016-12-20:
All seems to run just fine, but randomly I'm receiving this message when I start my python application

SystemError: DBus cannot connect to the OMXPlayer process

if I kill the process and run it again it runs just fine.

Is there any way to catch the dbus connection error, so that I can kill the script and let my watchdog run it again ?

Thanks

willprice, 2016-12-20:
Have you tried a `try except SystemError` [block](https://docs.python.org/2.7/tutorial/errors.html)?

It's odd that's happening, can you reproduce this in a minimal example? Perhaps we can figure out what's going wrong.

nsuva, 2016-12-20:
Thanks for your fast answer Will

My code it's similar to issue #61
Player initialization is the same and I'm grabbing input from GPIO or touch thru pygame.

I have a SystemErtor block but I don't know if it is well written because when the issue happens it doesn't enter this block.
Other weird thing is that even after the error prints on screen the movie start playing, but can't control it.
I'm not at computer right now but if you need I can send you the code to analyze, and any further test just let me know.
Thanks one more time and congrats for the good work you made with this.

willprice, 2016-12-20:
I'm pretty sure the DBus connection isn't being initialised, have a look at this line here: https://github.com/willprice/python-omxplayer-wrapper/blob/280b32f7a546ea38bb5057289aa83b47312cdfd7/omxplayer/player.py#L128

That's where the exception is thrown, as you can see we try to initialise a connection to DBus, could you have a look at what files are left in /tmp/ with `omxplayer` in their name before and after you run your script.

nsuva, 2016-12-21:
I Will, 
After a faulty start inside /tmp I get the following: 

pi@DIGITAL01:/tmp $ cat omxplayerdbus.pi
unix:abstract=/tmp/dbus-aOFZyO99B7,guid=cb75f4c6e8f20c24dcdbc2ab585a4988
pi@iDIGITAL01:/tmp $ cat omxplayerdbus.pi.pid
1171

After this I've killed the processes and run it again and all works fine. It happens randomly
_____________________________________________________________________

**The errors after python call are this :**

pi@DIGITAL01:~/media $ Traceback (most recent call last):
  File "player.py", line 26, in <module>
    player = OMXPlayer("/var/www/html/reglete/movie.mp4", args=['--no-osd','--lo    op','--no-keys','--aspect-mode','fill'], pause=True)
  File "build/bdist.linux-armv6l/egg/omxplayer/player.py", line 75, in __init__
    if GPIO.input(PIR_PIN) and player.position() > loop_start and time.time() >     t_end:
  File "build/bdist.linux-armv6l/egg/omxplayer/player.py", line 156, in load

  File "build/bdist.linux-armv6l/egg/omxplayer/player.py", line 82, in _load_sou    rce

  File "build/bdist.linux-armv6l/egg/omxplayer/player.py", line 119, in _setup_d    bus_connection

  File "build/bdist.linux-armv6l/egg/omxplayer/dbus_connection.py", line 23, in     __init__
  File "build/bdist.linux-armv6l/egg/omxplayer/dbus_connection.py", line 27, in     _create_connection
  File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 122, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not recei    ve a reply. Possible causes include: the remote application did not send a reply    , the message bus security policy blocked the reply, the reply timeout expired,     or the network connection was broken.

clickworkorange, 2017-03-29:
I have the same issue; OMXPlayer is launched and the load (paused) method is triggered by GPIO - toggling the input works successfully sometimes, sometimes I get "SystemError: DBus cannot connect to the OMXPlayer process". This is on latest Rasbian (2017-03-02) and OMXPlayer dfea8c9 (2016-09-23). Odd thing is, the exact same code runs perfectly on a Rasbian 2016-11-25 set-up, though I no longer have access to that system so cannot make a proper comparison.

3djake, 2017-08-23:
I had similar issues and what fixed it for me was
`rm /tmp/omxplayer*`

I think it had files from before I upgraded my code python3.x

bhong2002, 2017-08-25:
hello :)

i have this code:

------------------------------------------------------------------------------------------------------------
`from time import sleep
import time
from omxplayer import OMXPlayer

SCREEN_SAVER = '/home/pi/Videos/clip.mp4'
`
try:
    player = OMXPlayer(SCREEN_SAVER, args=['--dbus_name', 'org.mpris.MediaPlayer2.omxplayer0', '--win', '85 55 640 280', '-o', 'both',
                                       '--layer', '2', '--vol', '-1500'], pause=True)

    screensaver = OMXPlayer(SCREEN_SAVER, args=['--dbus_name', 'org.mpris.MediaPlayer2.omxplayer1', '--win', '50 50 810 420', '-o', 'both', '--loop', 
                                            '--no-osd', '--layer', '1', '--vol', '-3060'], pause=True)
`except Exception as err:
    print("****** ERROR: ******", str(err))`
``

and i got this error:
``

****** ERROR: ****** DBus cannot connect to the OMXPlayer process
Traceback (most recent call last):
  File "/usr/share/", line 9, in <module>
    '--layer', '2', '--vol', '-1500'], pause=True)
  File "/usr/local/lib/python3.4/dist-packages/omxplayer_wrapper-0.2.3-py3.4.egg/omxplayer/player.py", line 81, in __init__
    self.load(source, pause=pause)
  File "/usr/local/lib/python3.4/dist-packages/omxplayer_wrapper-0.2.3-py3.4.egg/omxplayer/player.py", line 162, in load
    self._load_source(source)
  File "/usr/local/lib/python3.4/dist-packages/omxplayer_wrapper-0.2.3-py3.4.egg/omxplayer/player.py", line 88, in _load_source
    self._connection = self._setup_dbus_connection(self._Connection, self._bus_address_finder)
  File "/usr/local/lib/python3.4/dist-packages/omxplayer_wrapper-0.2.3-py3.4.egg/omxplayer/player.py", line 134, in _setup_dbus_connection
    raise SystemError('DBus cannot connect to the OMXPlayer process')
SystemError: DBus cannot connect to the OMXPlayer process
>>> 

``

my /tmp got this:
``
pi@:~ $ ls /tmp/omx*
/tmp/omxplayerdbus.pi  /tmp/omxplayerdbus.pi.pid
pi@:~ $ cat /tmp/omxplayerdbus.pi
unix:abstract=/tmp/dbus-US9BqIMmnC,guid=62e45fd51b635c9da0f1e557599fd987
pi@:~ $ cat /tmp/omxplayerdbus.pi.pid
4289
pi@:~ $ 
``

my system:
``
raspberry pi3
4.9.40-v7+ #1022 SMP Sun Jul 30 11:16:10 BST 2017 armv7l GNU/Linux
Python 3.4.2
python3-dbus - 1.2.0-2+b1
omxplayer - 061425a
``

i didn't get 2nd dbus named --dbus_name', 'org.mpris.MediaPlayer2.omxplayer1

i also tried to remove the --dbus-name on the "player" and it still got the same error
 "player = OMXPlayer(SCREEN_SAVER, args=['--win', '85 55 640 280', '-o', 'both',
                                       '--layer', '2', '--vol', '-1500'], pause=True)"

thanks :) 

``

3djake, 2017-08-25:
Hello,

maybe try
```['--win', '50',  '50',  '810',  '420']```

bhong2002, 2017-08-25:
i've try that before . . .  even i remove that.... i also remove --dbus-nae on both .. and it will play but this error appears:

dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.5 was not provided by any .service files

bhong2002, 2017-08-25:
this my last result tests on shell with multiple loading attempt

Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "copyright", "credits" or "license()" for more information.
>>> from omxplayer import OMXPlayer
>>> player = OMXPlayer('/home/pi/Videos/clip.mp4', args=['--win', '85 55 640 280', '-o', 'both', '--layer', '2', '--vol', '-1500'], pause=True)
>>> player.load('/home/pi/Videos/clip.mp4', pause=True)
>>> player.load('/home/pi/Videos/clip.mp4', pause=True)
>>> player.load('/home/pi/Videos/clip.mp4', pause=True)
>>> player.load('/home/pi/Videos/clip.mp4', pause=True)
>>> player.load('/home/pi/Videos/clip.mp4', pause=True)
>>> player.load('/home/pi/Videos/clip.mp4', pause=True)
>>> player.load('/home/pi/Videos/clip.mp4', pause=True)
Traceback (most recent call last):
  File "<pyshell#8>", line 1, in <module>
    player.load('/home/pi/Videos/clip.mp4', pause=True)
  File "/usr/local/lib/python3.4/dist-packages/omxplayer_wrapper-0.2.3-py3.4.egg/omxplayer/player.py", line 162, in load
    self._load_source(source)
  File "/usr/local/lib/python3.4/dist-packages/omxplayer_wrapper-0.2.3-py3.4.egg/omxplayer/player.py", line 88, in _load_source
    self._connection = self._setup_dbus_connection(self._Connection, self._bus_address_finder)
  File "/usr/local/lib/python3.4/dist-packages/omxplayer_wrapper-0.2.3-py3.4.egg/omxplayer/player.py", line 134, in _setup_dbus_connection
    raise SystemError('DBus cannot connect to the OMXPlayer process')
SystemError: DBus cannot connect to the OMXPlayer process
>>>

3djake, 2017-08-26:
does it play the file when you run the command with the same arguments from the terminal?

also have you tried to removing pause?

bhong2002, 2017-08-26:
yes it is normally played.... at first.. then after another .load() as i posted before .....another video file. the error will come.... and i can't play it again

bhong2002, 2017-08-26:
viewing On task manager (a "top command on terminal").. i've seen a lot of dbus-daemon with different pid.. 
..

jehutting, 2017-08-26:
@bhong2002 Assuming you are running an up-to-date pyhton-omxplayer wrapper, **the wrapper OMXPlayer has an argument dbus_name** which you need to use when you want to name your real OMXPlayer.

You are setting the name through the wrapper OMXPlayer **args** argument. The real OMXPlayer gets that name (in your case e.g. org.mpris.MediaPlayer2.omxplayer**0**). The wrapper, however, is not aware of your rename request. So it tries to access the real OMXPlayer through the default name org.mpris.MediaPlayer2.omxplayer. This mismatch makes the wrapper unable to connect to the real OMXPlayer (process) resulting in the _SystemError: DBus cannot connect to the OMXPlayer process_.

Try this:
```
from time import sleep
import time
from omxplayer import OMXPlayer

SCREEN_SAVER = '/home/pi/Videos/clip.mp4'

try:
    player = OMXPlayer(SCREEN_SAVER, args=['--win', '85 55 640 280', 
                            '-o', 'both', '--layer', '2', '--vol', '-1500'], 
                            dbus_name='org.mpris.MediaPlayer2.omxplayer0',
                            pause=True)

    screensaver = OMXPlayer(SCREEN_SAVER, args=[
                            '--win', '50 50 810 420', '-o', 'both', '--loop', '--no-osd', '--layer', '1',
                            '--vol', '-3060'], 
                            dbus_name='org.mpris.MediaPlayer2.omxplayer1',
                            pause=True)

except Exception as err: 
    print("****** ERROR: ******", str(err))
```

bhong2002, 2017-08-27:
hello sir @jehutting   :) . . 

trying the above code you posted .. .. got this . . .

****** ERROR: ****** __init__() got an unexpected keyword argument 'dbus_name'

thanks :)

jehutting, 2017-08-27:
Did you update to the latest version of the python-omxplayer-wrapper?

bhong2002, 2017-08-27:
the omxplayer-wrapper version i am using is 0.2.3 ..  is there any latest version?

thanks :)

jehutting, 2017-08-27:
See 60fa8fecd908a8c66e398c85ffdb671420fee687 for the dbus_name support. The version **number** (in setup.py) is not yet incremented, so it doesn't reflect the change. I think you need to do a new ```git pull``` and run the setup.

bhong2002, 2017-08-27:
thanks a lot sir @jehutting :+1:  . . . pulled done ...  it works... i wiil post again if i got another error..

thanks again :)

closed 2017-09-26 by willprice

DanTheMan2000, 2020-08-18:
Pull request that is related in popcornmix/omxplayer
https://github.com/popcornmix/omxplayer/pull/620