willprice/python-omxplayer-wrapper #69: OMXplayer does not start when specifying the "-o hdmi" option

congoblue, 2017-01-06:
I hope I am not doing something really stupid here... 
When starting OMXPlayer using the wrapper, if the "-o hdmi" option is included in the args list then OMXPlayer apparently does not start. Returns SystemError: DBus cannot connect to the OMXPlayer process.
If OMXPlayer is launched from the commandline then "-o hdmi" works correctly

## Problem reproduction:
from omxplayer import OMXPlayer
player=OMXPlayer('test.mp4', args=['--no-osd', '-o hdmi', '-b'])
#testing, the line below works, but no audio on hdmi
#player=OMXPlayer('test.mp4', args=['--no-osd', '-b'])

willprice, 2017-01-06:
The args need to be split as if the shell split them: `args=['--no-osd', '-o', 'hdmi', '-b']`.

If you don't want to do this manually you can just pass the string you'd usually to the shell e.g. `'--no-osd -o hdmi -b'` to [`shlex.split`](https://docs.python.org/2/library/shlex.html) which will result in the array above.

congoblue, 2017-01-06:
Ah, so it was something stupid, thanks for the quick response. This library is a very useful improvement to omxplayer, thanks for the work on it.

willprice, 2017-01-06:
No problem @congoblue, in fact this raises quite a serious bug IMO, we should raise an exception if OMXPlayer quits out as soon as it starts, the DBus exception is very unhelpful, I'll add this as a bug.

congoblue, 2017-01-07:
OK thank you. Would this also deal with the situation if OMXPlayer fails to play the file for some reason, at the moment the player=OMXPlayer setup call appears to succeed but the next time you try to do something with player you get "DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply" - but I can't see a way of telling what happened.

willprice, 2017-01-07:
Yeah that sounds like the same problem, anything that would cause the player to exit and not play a video would benefit from this change.

I'm closing this as I'm tracking it in #70 now (anything causing immediate death)

closed 2017-01-07 by willprice