willprice/python-omxplayer-wrapper #71: Issue playing clip

davemccall, 2017-01-09:
# Issue Report
## Description
I'm trying to play a clip of a local mp4 file. Though several tries, I could only get it to work if I queried the duration. I'm happy to use this as a workaround, just not clear why it is necessary.

Without the player.duration(), if I set_position then the video will not play. With the player.duration() eveyrthing works perfectly.

## Problem reproduction

def playMedia(filename="demo.mp4", duration=0, position=0):
	player = OMXPlayer("/home/pi/bionic/media/" + filename, args=["--no-osd"])
	player.set_aspect_mode("fill")
	if position > 0:
		player.set_position(position)
	player.duration() # this only works if this line is here
	if duration == 0:
		duration = player.duration() - position
	player.play()
	time.sleep(duration)
	player.quit()
	return True

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.1.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0` |
| `python` (`python --version`) | `3.4.2` | (modified urllib references and long() cast)
| `omxplayer` (`omxplayer --version`) | `dfea8c9` |

willprice, 2017-01-09:
That's odd.

What do you mean by "get it to work", can you outline the expected behaviour and the observed behaviour before and after adding in the additional line.

Thanks

davemccall, 2017-01-11:
If I remove the line in question, the video simply won't play. There is no error, it just does not play. If I remove the line in question and do not set_position, the video will play (from the beginning). I just can't seem to play the video from a set position without querying the duration.

willprice, 2017-09-26:
Could not reproduce from 1d9119c67479155d04c6bee198dd27f2db3ce322

closed 2017-09-26 by willprice