willprice/python-omxplayer-wrapper #72: Use multiple instance of player

nicoske, 2017-01-31:
# Issue Report
## Description
I'm trying to use the wrapper to show multiple instance of live streams.

## Problem reproduction
Here's a sample of code:
`from omxplayer import OMXPlayer
from time import sleep

stream1l = 'rtmp://stream.lan:1935/mv/1.sdp'
stream2 = 'rtmp://stream.lan:1935/mv/2.sdp'

player1 = OMXPlayer(stream1, args=['--dbus_name=player1'])
player1.play()
player1.set_video_pos(0,0,960,540)
player2 = OMXPlayer(stream2, args=['--dbus_name=player2'])
player2.play()
player2.set_video_pos(960,0,1920,540)
`

Only 1 omxplayer instance start and the video is show in fullscreen (video_pos not working).
'
process 11334: arguments to dbus_bus_request_name() were incorrect, assertion "_dbus_check_is_valid_bus_name (name)" failed in file ../../dbus/dbus-bus.c line 1122.
This is normally a bug in some application using the D-Bus library.
Traceback (most recent call last):
  File "omx.py", line 9, in <module>
    player1 = OMXPlayer(berl, args=['--dbus_name=player1'])
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 75, in __init__
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 156, in load
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 82, in _load_source
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 128, in _setup_dbus_connection
SystemError: DBus cannot connect to the OMXPlayer process
'

## Environment details
* OS (`lsb_release -a`):
```
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0-2+b1` |
| `python` (`python --version`) | `2.7.9` |
| `omxplayer` (`omxplayer --version`) | `6c90c75` |

willprice, 2017-02-07:
I've never tried doing anything like this. I wonder whether those are valid DBus names? Usually the dbus name will be `org.mpris.MediaPlayer2.omxplayer`, have you tried `org.mpris.MediaPlayer2.omxplayer{1,2}` instead? 

It's likely that the `bus_finder` object will fail in this scenario as well, as far as I can remember it implicitly assumes only a single player. You'd probably want to inject a custom bus_finder pointing at the correct files made in /tmp

wmodes, 2017-02-11:
@nicoske, did you ever have luck creating multiple simulatenous instance of the player?

nicoske, 2017-02-16:
@willprice I tried to use `org.mpris.MediaPlayer2.omxplayer1` and `org.mpris.MediaPlayer2.omxplayer1` but I face the same problem. 

@wmodes Yes I'm able to manually start 4 instance of omxplayer like this:

`
omxplayer rtmp://streamsrv/stream1.sdp --win '960 0 1920 540'
omxplayer rtmp://streamsrv/stream2.sdp --win '0 0 960 540'
omxplayer rtmp://streamsrv/stream3.sdp --win '960 540 1920 1080'
omxplayer rtmp://streamsrv/stream4.sdp --win '0 540 960 1080'
`

What I would like to achieve is to change window position while the instance are running.

willprice, 2017-02-16:
Hi @nicoske,

Interesting... so I wonder how omxplayer decides what names to use on DBus? What do you have in /tmp/ named something like `omxplayer*`? If we can figure out how omxplayer names itself on DBus when multiple instances are running, we can update `bus_finder` to support multiple players, currently it is dumb and just uses the first `omxplayer.blah.*` file it finds in /tmp and parses the DBus address from that.

nicoske, 2017-02-24:
Hi @willprice 

Strangely there's only  2 files in /tmp

`-rw-r--r--  1 pi pi   73 Jan 23 14:17 omxplayerdbus.pi
-rw-r--r--  1 pi pi    4 Jan 23 14:17 omxplayerdbus.pi.pid
`
omxplayerdbus.pi contains:

'unix:abstract=/tmp/dbus-eExZIacLOG,guid=b6be7ba8aa90a414b0010f775886106d'

If I specify a different dbus name for each of the instance (org.mpris.MediaPlayer2.omxplayer1 & org.mpris.MediaPlayer2.omxplayer2) there's still only 2 files in /tmp (pid + dbus). The second player instance does not even overwrite those files, they are from the first instance running.

So I assume it works but it is not designed to run multiple instance...

Thanks anyway

nicoske, 2017-02-24:
Just to confirm that omxplayer is re-using the same abstract socket name. With 2 instances using different dbus_name here's the result of a netstat:

`unix  3      [ ]         STREAM     CONNECTED     38498    27093/dbus-daemon   @/tmp/dbus-TXwSWoO7c0`
`unix  3      [ ]         STREAM     CONNECTED     38522    27093/dbus-daemon   @/tmp/dbus-TXwSWoO7c0`

willprice, 2017-02-24:
Ah, clever, I didn't know that was possible. So just to confirm my
understanding: you're saying that all omxplayer processes reuse the same DBus
abstract socket name and hence there is only one set of
`{dbus<name>,dbus<name.pid} `files in `/tmp`?

  

I wonder what happens when the first omxplayer process who initialises the
DBus interfaces dies, do the other omxplayer processes (started after the
first) still work? Or does DBus communication then fail? (I want to know
whether the DBus socket is closed when the initialising process dies, or
whether it sticks around until all omxplayer instances using the socket die)

> Just to confirm that omxplayer is re-using the same abstract socket name.
With 2 instances using different dbus_name here's the result of a netstat:

> `unix 3 [ ] STREAM CONNECTED 38498 27093/dbus-daemon @/tmp/dbus-TXwSWoO7c0`  
`unix 3 [ ] STREAM CONNECTED 38522 27093/dbus-daemon @/tmp/dbus-TXwSWoO7c0`

/python-omxplayer-wrapper/issues/72#issuecomment-282233948&r=cmVwbHkrMDAwZWJlM
2RiYTJiMWIwMmE2MzNkZjE3YTcxYjM1NzJmOTUzN2E4ZDlkOTM0M2JjOTJjZjAwMDAwMDAxMTRjN2I
xMDQ5MmExNjljZTBjMmMzNTg2QHJlcGx5LmdpdGh1Yi5jb20=), or [mute the
JUCFn21LkwvWks5rfpUEgaJpZM4LyUfm&r=cmVwbHkrMDAwZWJlM2RiYTJiMWIwMmE2MzNkZjE3YTc
xYjM1NzJmOTUzN2E4ZDlkOTM0M2JjOTJjZjAwMDAwMDAxMTRjN2IxMDQ5MmExNjljZTBjMmMzNTg2Q
HJlcGx5LmdpdGh1Yi5jb20=).![](https://github.com/notifications/beacon/AA6-PRUPH
2Y3ECtKNExT9JS_g8uoegy_ks5rfpUEgaJpZM4LyUfm.gif)

longagofaraway, 2017-07-17:
It's easy: pass to omxplayer arguments `--dbus younameit` then in dbus_connection.py in self._bus.get_object use that name.
omxplayer overwrites /tmp/omxplayerdbus.${USER} every time it starts so everything is working

longagofaraway, 2017-07-17:
I can upload the changes but don't have permission.

willprice, 2017-07-19:
Hey @longagofaraway, I'd appreciate a pull request with your changes.

longagofaraway, 2017-07-20:
I had to remove files from /tmp because I killed dbus process associated with them. If you don't do anything like that, everything should be fine.

And I still have issues with pushing my branch here.

remote: Permission to willprice/python-omxplayer-wrapper.git denied to longagofaraway.
fatal: unable to access 'https://github.com/willprice/python-omxplayer-wrapper.git/': The requested URL returned error: 403

longagofaraway, 2017-07-20:
I got it, I have to fork it.

nicoske, 2017-07-31:
Hey @longagofaraway thanks for the fork!
It works till the 3rd instance, the 4th one fail with dbus timeout. any idea why?

```
Traceback (most recent call last):
  File "omx.py", line 19, in <module>
    player4.set_video_pos(960,540,1920,1080)
  File "<decorator-gen-27>", line 2, in set_video_pos
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 151, in wrapped
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 403, in set_video_pos
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
```

This is the python code:

```
from omxplayer import OMXPlayer
from time import sleep

stream1= 'rtsp://10.0.5.170:554/stream'
stream2 = 'rtsp://10.0.13.250:554/stream'
stream3 = 'rtsp://10.0.17.212:554/stream'
stream4 = 'rtsp://10.0.8.133:554/stream'

player1 = OMXPlayer(stream1)
player1.set_video_pos(0,0,960,540)

player2 = OMXPlayer(stream2, dbus_name='org.mpris.MediaPlayer2.omxplayer1')
player2.set_video_pos(960,0,1920,540)

player3 = OMXPlayer(stream3, dbus_name='org.mpris.MediaPlayer2.omxplayer2')
player3.set_video_pos(0,540,960,1080)

player4 = OMXPlayer(stream4, dbus_name='org.mpris.MediaPlayer2.omxplayer3')
player4.set_video_pos(960,540,1920,1080)

```

jehutting, 2017-08-05:
Works with 'normal' videos. 

How big is your `gpu_mem` in /boot/config.txt?

If it still fails for let's say 128MB, add ```args=['-g']``` to the 4th player to generate an omxplayer.log.
Maybe the logging shows why omxplayer is not responding on the dbus command.

willprice, 2017-09-26:
Probably due to dbus issues similar to #65.

closed 2017-09-26 by willprice

Akwariom, 2021-06-18:
Hi. 
I am using two players, on two different screens on a rpi4, 
Works fine until I try to load a new video on the second video player. 
The load() function works fine with the first player, crashes the second. 
I also tried to quit() them both, and recreate both players, same issue : the player1 reopens, the second crashes with the "DBus cannot connect to the OMXPlayer process". Any idea what cause this ? 
Thanks