willprice/python-omxplayer-wrapper #73: Player never returns stopped

wmodes, 2017-02-07:
On the Raspi build version 0.2.0, python-omxplayer-wrapper never returns 'stopped' when playing a video after it reaches the end. Similarly play_sync() never returns.

    from omxplayer import OMXPlayer

       player = OMXPlayer(filename)
        player.play()
        while player.is_playing():
            print "status:", player.playback_status(),
            print "position:", player.position()
        print "Done."

Am I missing something?

willprice, 2017-02-07:
Can you please fill in the issue template with all the versions you're using (https://github.com/willprice/python-omxplayer-wrapper/blob/develop/.github/ISSUE_TEMPLATE.md)

If you want to play multiple videos you'll need to create another omxplayer object after the first has died.

wmodes, 2017-02-07:
Done.

wmodes, 2017-02-07:
I only want to play one video at a time. Is it not possible to load a new video file into the existing player? if not, since the instantiation requires a filename, what is the purpose of the ``player.load()`` method?

The challenge is that there is a half second pause as a new OMXplayer is created, preventing seamless video transitions. 

In any case, does the ``player.is_playing()`` ever return anything other than ``playing``? And I presume that when the player gets to the end of the file it is not supposed to freeze. 

Today, after I waited a bit after the player froze, I got the following:

''Traceback (most recent call last):
  File "omx-test.py", line 108, in <module>
    main()
  File "omx-test.py", line 89, in main
    play_film(player, filename=choice(trans_films), duration=1)
  File "omx-test.py", line 43, in play_film
    full_length = player.duration()
  File "<decorator-gen-17>", line 2, in duration
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 141, in wrapped
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 305, in duration
  File "<decorator-gen-16>", line 2, in _duration_us
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 141, in wrapped
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 297, in _duration_us
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)''

willprice, 2017-02-08:
`load` is used to kill the current player and fork a new one, once the initial player has died, it *should* still be able to fork a new process and open a new video, but if that's not the case, please file another bug report with this issue.

willprice, 2017-02-08:
If you want to play multiple videos, concatenate them into one big video file, then skip around that file (that way you only have to use one instance of omxplayer and you avoid the delay of forking another omxplayer process)

wmodes, 2017-02-09:
Other than one big file (which might work, thanks), do you have any other suggestions for playing videos seamlessly?

For instance, it is possible to start a new player, get it running, and then get a callback when the first video completes?

wmodes, 2017-02-09:
When the load() method forks a new player, does the old one ever die? Is there a way to send it a signal to die?

willprice, 2017-02-11:
One big file works best due to the cost of launching new omxplayer processes (which takes ~1 second). I've done this and the result was far superior to launching new processes.

Sadly I'm unaware of a way to find out when the video has finished (although we do monitor the omxplayer process which dies when the video has finished-perhaps we can hook in here to get an `omxplayer_exited` callback).

willprice, 2017-02-11:
> When the load() method forks a new player, does the old one ever die? Is there a way to send it a signal to die?

We try and kill the player first, have a look at the source code for `load`

Gajillion, 2017-08-06:
> (although we do monitor the omxplayer process which dies when the video has finished-perhaps we can hook in here to get an omxplayer_exited callback).

Was this ever done? I'm running into the same issue wmodes is. I need to be able to figure out when the video is finished so I can start up the next one. I actually don't need seemless playback.

bhong2002, 2017-09-07:
hi :),

>(although we do monitor the omxplayer process which dies when the video has finished-perhaps we can hook in here to get an omxplayer_exited callback). 

bringin' this up.... same issue here

thanks :)

willprice, 2017-09-07:
This may be possible but

Leocoe, 2018-04-12:
Any news on whether/when player.stopEvent will be triggered on playback completion?
By the way, awesome project =)

Landfishy, 2018-10-14:
Yeah, has there been any progress on this? This is literally the only thing I need out of this package. I'm trying desperately to figure out of theres a way to check the existing timestamp on the  video and compared it to video length as a substitute, but I don't know if that is possible either.

Tastyep, 2018-10-29:
Bumping this, I saw that a PR adding the callback got merged on the develop branch.
What is blocking for this to make it to the master branch?

Landfishy, 2018-11-01:
I'm still trying to sort this out, though in the meantime I've been cross examining omxplayer-wrapper, (which willprice also worked on) which already features this functionality (and it works!).

willprice, 2018-11-01:
You can install the `develop` branch by `pip install git+https://github.com/willprice/python-omxplayer-wrapper.git@develop`.
I may have a look at this on the weekend, but I'm busy with deadlines until late November so no promises.

willprice, 2018-11-03:
v0.3.0 is out on PyPI with the `exitEvent`. Closing this, if there are any issues with the `exitEvent` please open another issue.

closed 2018-11-03 by willprice