willprice/python-omxplayer-wrapper #74: cannot set --vol with args[]?

artaslove, 2017-02-08:
# Issue Report
## Description
omxplayer.bin doesn't handle the initial volume argument. When play() is invoked, the video plays at full volume for about a second before changing to the volume set with set_volume(). 

When launching omxplayer from the command line, the argument '--vol -6000' works as expected. 

## Problem reproduction

$ omxplayer something.mp4 --vol -6000
# works as expected

>>> video = omxplayer.OMXPlayer('something.mp4')
>>> video.set_volume(-6000)
>>> video.play()
# video plays with first second at full volume, then is silent

>>> video = omxplayer.OMXPlayer('something.mp4', args=['--vol -6000'])
>>> video.set_volume(-6000)
>>> video.play()
/usr/bin/omxplayer.bin: unrecognized option '--vol -6000'

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.1.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0-2+b1` |
| `python` (`python --version`) | `3.4.2` |
| `omxplayer` (`omxplayer --version`) | `dfea8c9` |

-------

# Feature Request
## Description
Allow setting of initial volume

## Example use case
Playing a video with consistent volume

willprice, 2017-02-08:
Hey,

Thanks for filling up the bug report form!

You've not split the arguments as the shell would, try with `args=['--vol', '-6000']`, this is about the 3rd time someone has reported this as a bug indicating the docs aren't clear enough that you must split the args as the shell would (or alternatively use `shlex` to do it for you).

I'll close this (reopen if my fix above doesn't work) and open a new issue to improve the docs.

Thanks

closed 2017-02-08 by willprice

artaslove, 2017-02-08:
Lol! Thanks so much. That works perfectly.

willprice, 2017-02-08:
No problem, thanks for reporting, I count this as a docs bug ;)  

/python-omxplayer-wrapper/issues/74#issuecomment-278463380&r=cmVwbHkrMDAwZWJlM
2RjZjVlYWI5MWMxYWNkOGEzM2QxN2RlYmQ3NzhjM2QzNTJkMjQ2YjVkOTJjZjAwMDAwMDAxMTRiMzR
iZmI5MmExNjljZTBjNGMzOGI5QHJlcGx5LmdpdGh1Yi5jb20=), or [mute the
-PTTa5S4jjeNw0McryHW3CfX-FeQ6ks5rai_7gaJpZM4L7WbS&r=cmVwbHkrMDAwZWJlM2RjZjVlYW
I5MWMxYWNkOGEzM2QxN2RlYmQ3NzhjM2QzNTJkMjQ2YjVkOTJjZjAwMDAwMDAxMTRiMzRiZmI5MmEx
NjljZTBjNGMzOGI5QHJlcGx5LmdpdGh1Yi5jb20=).![](https://github.com/notifications
/beacon/AA6-PUyqgIYwdOmJo4vUX5ogs4dYdpxsks5rai_7gaJpZM4L7WbS.gif)