willprice/python-omxplayer-wrapper #77: no module named 'omxplayer' in Python 3

ash73, 2017-02-12:
# Issue Report
## Description

When using omxplayer-wrapper with Python 3 I get the error:
No module named 'omxplayer'

I saw a previous (closed) thread that suggested:
sudo apt-get install python3-dbus

But it makes no difference for me, I still get the same error.

## Problem reproduction

git clone https://github.com/willprice/python-omxplayer-wrapper
sudo apt-get install python-dbus
sudo apt-get install python3-dbus
cd python-omxplayer-wrapper
sudo python setup.py install
reboot

Then load and run the sample in Python 3.

## Environment details

Raspbian on Pi2.

TIA.

willprice, 2017-02-12:
Python 3 isn't officially supported, although some people have fiddled with a few lines and got it to work, you're invoking `setup.py` with `python` which on raspbian is `python2` not `python3`, so you should be running `python3 setup.py install` instead.

Have a look at #66 for more details

ash73, 2017-02-12:
Thanks, took a while as a noob to decipher #66, full sequence is:

git clone https://github.com/willprice/python-omxplayer-wrapper
sudo apt-get install python3-dbus
cd python-omxplayer-wrapper
sudo nano omxplayer/player.py

search for import urlparse change to 
  import urllib.parse

search for def _setup_omxplayer_process change second line to
  source_url = urllib.parse.urlsplit(source)

search for return long(self._get_properties_interface().Duration())
  replace long(  ) with int(  )

sudo python3 setup.py install

willprice, 2017-02-12:
Cool, thanks for providing the steps in sequence for someone else looking how to do this. When I eventually get around to making this compatible for both py2 and py3 I'll refer back to this.

willprice, 2017-04-17:
Python 3 support has been implemented in https://github.com/willprice/python-omxplayer-wrapper/tree/python_2_3_support

I need beta testers to see whether this works with their applications. Ideally please also run the smoke tests and report any issues.

willprice, 2017-04-18:
Completed in 809614b1e373dc18a89ef45680f8cc508a6f3d8a

If problems arise on py3 and not py2 or vica versa, please do file an issue.

closed 2017-04-18 by willprice

SPFXUNLIMITED, 2017-10-25:
No module named 'omxplayer'