willprice/python-omxplayer-wrapper #79: Any working examples using OMXPlayer-wrapper with GPIO buttons?

darrell24015, 2017-03-31:
## README
If you're looking for **help** on how to use the library head over to the
[Raspberry Pi forums](https://www.raspberrypi.org/forums/viewforum.php?f=32&sid=f1a4513e3e137272da39dbb11089e077)

If you're reporting an issue or requesting a feature, read on! Pick one of the
following templates depending on your issue type.

------
Delete above and including this line.

# Issue Report
## Description
I am not getting any results trying to start/pause/restart a video with GPIO buttons. The video only plays with no further ability to control it.

## Problem reproduction
Sample code thus far:

`
from omxplayer import OMXPlayer
from time import sleep
import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(True)

myButton = 4

GPIO.setup(myButton, GPIO.IN, GPIO.PUD_UP)

def getButtonPress():
   while 1:
      if GPIO.input(myButton) == GPIO.Low:
         return True
         break

video = '/home/pi/path_to_video.mp4'

player = OMXPlayer(video)

try:
   if getButtonPress() == True:
      player.play()

except KeyboardInterrupt:
   print('Closing Player')
   GPIO.cleanup()
   player.quit()
`

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.1.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0` |
| `python` (`python --version`) | `2.7.9` |
| `omxplayer` (`omxplayer --version`) | `6c90c75` |

-------

# Feature Request
## Description
Would like to be able to pause and then reset the video with a second button press, or stop the video with the KeyboardInterrupt

## Example use case
An example of how the feature you request would be used, demonstrating it's
utility (and why the current solution doesn't solve your problem).

closed 2017-04-20 by darrell24015

willprice, 2017-04-20:
Did you manage to produce a working example @darrell24015? If so I'm sure others would appreciate it if you could share it :)

JE-Random-Tech, 2019-09-22:
I have an example with GPIO Pins which you can see [here](https://www.youtube.com/watch?v=a476_6VVs5Y). Hope this helps someone.