willprice/python-omxplayer-wrapper #81: ImportError: cannot import name OMXPlayer, python2 & 3, RPI2, wrapper 0.2.1

Octane70, 2017-04-19:
# Issue Report
## Description
getting a "ImportError: cannot import name OMXPlayer" using either  Python2 or 3 on RaspberryPi 2

#Install Python
used $ sudo apt-get install python-dbus and is installed
then sudo python setup.py install

#Install Python3
used $ sudo apt-get install python3-dbus and is installed
then sudo python3 setup.py install

## Problem reproduction
from omxplayer import OMXPlayer
from time import sleep

file_path_or_url = '/home/pi/test_media_1.mp4'

player = OMXPlayer(file_path_or_url)

# The player will initially be paused

player.play()
sleep(5)
player.pause()

# Kill the `omxplayer` process gracefully.
player.quit()

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.1` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0-2` |
| `python` (`python --version`) | `2.7.9` & '3.4.2' |
| `omxplayer` (`omxplayer --version`) | `dfea8c9 ` |

willprice, 2017-04-19:
That's odd, I'd not expect this to happen, nor can I reproduce the issue under either python 3 or 2.

Can you give me the full output of the command `python setup.py install` for both python versions

willprice, 2017-04-19:
Can you also try installing it as a dev package using `python setup.py dist`, can you use it then? (It installs a single file `/usr/local/lib/pythonX/dist-packages/omxplayer-blah.egg-link` linking to the src directory).

Can you install and use it for PyPI using `pip`?

Octane70, 2017-04-19:
Hey Will , it seemed to have started working in both pythons, not sure what made it work. I just started adding code to a Tkinter Gui I Have been working on,  ill keep you posted on any other issues.

willprice, 2017-04-19:
Sweet, thanks for letting me know.

I've only very recently added python3 compatibility so I thought I managed to botch the release!

closed 2017-04-19 by willprice