willprice/python-omxplayer-wrapper #82: youtube-dl as source

varadig, 2017-04-26:
Can I use, and if I can, how to pass source as youtube link.
From cli I can play the video on this way:
`omxplayer $(youtube-dl -g https://www.youtube.com/watch?v=zAoWcs6d75Q)`
But with python-omxplayer-wrapper I got this error:
```
  File "/usr/local/lib/python2.7/dist-packages/omxplayer_wrapper-0.2.1-py2.7.egg/omxplayer/player.py", line 114, in _setup_omxplayer_process
    raise FileNotFoundError("Could not find: {}".format(source))
FileNotFoundError: Could not find: $(youtube-dl -g https://www.youtube.com/embed/yA-qwirVwbs)
```

There is any option to play youtube video with the wrapper, or there is any chance to add support to play it?

willprice, 2017-04-26:
Because python does not delegate to the shell when running omxplayer you cannot use shell construct slike `"$(youtube-dl -g ...)"`.

I'd create a named FIFO (https://docs.python.org/3/library/os.html#os.mkfifo), you can then invoke `youtube-dl` piping to the FIFO, pass then pass the FIFO path to omxplayer in place of a normal file.

varadig, 2017-04-26:
Thanks, i will check ;)

varadig, 2017-04-26:
Hmmm with commandline I can do the job, but with wrapper I thing I miss something.
Can you show me your example of the usage?

**python:**
```
import os

import thread
from threading import Thread
from time import sleep

from omxplayer import OMXPlayer

try:
    os.mkfifo("/home/pi/ytpy.fifo",0o666)
except:
    print 'fifo is exists'

def download():
    os.system("youtube-dl -o - QWY-Obragek > /home/pi/ytpy.fifo")

def play():
    try:
        player = OMXPlayer("/home/pi/ytpy.fifo")
        player.play()
    except:
        sleep(1)
        play()

thread = Thread(target = download)
thread.start()

thread2 = Thread(target = play)
thread2.start()
```

**cli:**
`youtube-dl -o - QWY-Obragek > /home/pi/ytpy.fifo & omxplayer -o local /home/pi/ytpy.fifo`

jambonmcyeah, 2017-07-25:
This is my solution:
```python
    import subprocess

    proc = subprocess.Popen(['youtube-dl','-f','best', '-g', url], stdout=subprocess.PIPE)
    realurl=proc.stdout.read()
    from omxplayer import OMXPlayer
    OMXPlayer(realurl.decode("utf-8", "strict")[:-1])
```

jehutting, 2017-08-05:
@jambonmcyeah Works like a charme. Thanks.

jambonmcyeah, 2017-08-10:
@jehutting You're welcome :)

closed 2017-09-26 by willprice