willprice/python-omxplayer-wrapper #83: Poor Performance

krajni, 2017-04-27:
First of all Thanks for your time and effort that you  spent with this project. 
Unfortunately all the Videos I tested on a Raspi 3 had not a good performance. The Videos stutter a lot and are not nice to look at. I tried various Resolutions (1080,720,480) but all of them were stuttering. I also tried  both Python 2 and 3 but both have that Issue. Is everyone else having that same problem? The Videos are played without stuttering when OMXPlayer is started directly from Terminal.

varadig, 2017-04-27:
Weird, I have a same good performance as omx cli. So maybe you need to check come other thing. How many MB you split to memory for the GPU?

krajni, 2017-04-27:
Thanks for the quick reply. 128 MB are reserved for GPU.

krajni, 2017-04-27:
And the code isn't fancy either. I'm just testing the performance right now with:
from omxplayer import OMXPlayer

player = OMXPlayer("/home/pi/signage/test.mov")
player.play()

varadig, 2017-04-27:
I'm using RaspbianLite with matchbox-window-manager and 256Mb GPU.
What about if use mp4 instead of mov?

krajni, 2017-04-27:
Thanks for the hint. Just tried mp4: Same result. And also upgraded first to 256MB and then to 512, but still not getting a better result.  I will try with a fresh OS.

willprice, 2017-09-26:
Do you get stuttering performance under omxplayer on the cli, if so then the bug resides in your setup or in omxplayer and not this wrapper.

willprice, 2017-09-26:
Closing unless someone can reproduce with a minimal working example.

closed 2017-09-26 by willprice