willprice/python-omxplayer-wrapper #84: Crash when file is finised

gchanteloube, 2017-05-19:
# Issue Report
## Description
Hi. When my sound file (simple mp3) is finished, I try to stop() and quit() the player, but I receive an exception : 

> dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

Thanks you!

## Problem reproduction
\>\>\> from omxplayer import OMXPlayer
\>\>\> player = OMXPlayer('test.mp3')
\>\>\> player.quit() or .stop() or .playback_status()

## Environment details
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.2 (develop or release` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0-2` |
| `python` (`python --version`) | `2.7.9` |
| `omxplayer` (`omxplayer --version`) | `dfea8c9` |

-------

willprice, 2017-05-20:
Hi @gchanteloube,

Thanks for the bug report. I don't have time right now to investigate this, but after my exams are finished ~3rd June I'll take a look :)

3djake, 2017-08-23:
Doesnt it kill itself when the file has finished playing? which is why you can no longer make dbus calls?

jehutting, 2017-08-26:
@3djake is correct as it is true that the (real) OMXPlayer process kills itself when it is finished and therefore one can no longer make dbus calls.

However it is strange that the NoReply dbus exception occurs as each dbus call (command) is guarded by a ```_check_player_is_active``` function. This function should lead to the error _Process is no longer alive, can't run command_.

Interesting part of @gchanteloube script is therefore the 
```player.quit() or player.stop() or player.playback_status()```
line.

This line results in
```
  File "/xxx/willprice/python-omxplayer-wrapper/omxplayer/player.py", line 56, in wrapped
    if self._process.poll() is None:
AttributeError: 'NoneType' object has no attribute 'poll'
```
as ```quit``` does (two times, one time is enough :) ```self._process = None```.

Removing the ```quit```from the line to have
```player.stop() or player.playback_status()```
results in
```
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
```
The ```stop``` command terminates the real OMXPlayer, but as the stop doesn't wait for the process to be terminated (in the same way as ```quit``` does), the ```playback_status``` can make the call _while_ the real OMXPlayer is terminating.

Maybe @gchanteloube can show more details of the script he is actually using.

willprice, 2017-09-26:
You should just use `quit` and avoid `stop` completely.

closed 2017-09-26 by willprice