willprice/python-omxplayer-wrapper #85: Omxplayer crashes with GPIO

balamsoto, 2017-05-24:
# Issue : crashes 
## Description:

using 3 gpio as input pull up to load movies makes the player crash after few pushbuttons

## Problem reproduction
after few gpio are grounded player crashes

code 
`

#!/usr/bin/env python2
import os.path
from time import sleep
import subprocess
import os
from omxplayer import OMXPlayer
vida = '/home/pi/Videos/testvids/6.mov'
vidb = '/home/pi/Videos/testvids/3.mov'
vidc = '/home/pi/Videos/testvids/t2.mp4'
default = '/home/pi/Videos/testvids/t1.mp4'

import RPi.GPIO as GPIO

#set up GPIO using BCM numbering

GPIO.setmode(GPIO.BCM)
#All Gpio's as input and pull up

GPIO.setup(2, GPIO.IN, pull_up_down = GPIO.PUD_UP)
GPIO.setup(3, GPIO.IN, pull_up_down = GPIO.PUD_UP)
GPIO.setup(4, GPIO.IN, pull_up_down = GPIO.PUD_UP)

player = OMXPlayer(default,args=['--no-osd','--blank'],)

while True:

      if GPIO.input(2) ==0:

           player.load(vida)
           print("gpio 2")
           player.play()
           #sleep(5)

      if (GPIO.input(3) == 0):

            player.load(vidb)
            print("gpio 3")
            player.play()
           # sleep(5)

      if (GPIO.input(4) == 0):
            player.load(vidc)
            print("gpio 4")
            player.play()
            #sleep(5)

GPIO.cleanup()

`

--------------------------------------
error  

> 
gpio 4
gpio 3
gpio 2
gpio 4
gpio 3
gpio 2
gpio 4
gpio 3
gpio 4
gpio 3
Traceback (most recent call last):
  File "mygpio.py", line 34, in <module>
    player.load(vida)
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 162, in load
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 88, in _load_source
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 134, in _setup_dbus_connection
SystemError: DBus cannot connect to the OMXPlayer process
> 

----------------------------------

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.1.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0` |
| `python` (`python --version`) | `2.7.9` |
| `omxplayer` (`omxplayer --version`) | `6c90c75` |

willprice, 2017-06-12:
How long are the videos you are loading in seconds?

Your program seems reasonable and is something that should be supported. I will have a look at replicated this with keypresses instead of GPIO and see whether I can reproduce the issue.

jehutting, 2017-06-12:
I have already done that (changing the GPIO by keyboardkeys):

Upon Player::load the function Player::_setup_dbus_connection is called to make a new OMXPlayer connection. When the connection to OMXPlayer fails, a self.tries is incremented (up to a max of 50).
Now when you do a second/third/etc. time load, the incrementation continues from where the previous load made the connection. So after an amount of loads (which varies depending on how fast the OMXPlayer connection is made) you run into the max value, resulting in the error.

The solution could be to place a `self.tries = 0` before the `while self.tries < 50)`; start from a refresh count.

willprice, 2017-06-12:
Could you provided the modified code as a minimal example?

Thanks for the suggestion, sounds good. I'll make the change and test it fixes.

jehutting, 2017-06-12:
This is the @balamsoto modified code (using keys in stead of GPIO)
```
#!/usr/bin/env python2

# Modified balamsoto code by jehutting
# See balamsoto's code at will price python-omxplayer-wrapper issue #85
# Original code uses GPIOs in stead of keyboard keys

import os.path
from time import sleep
import subprocess
import os
from omxplayer import OMXPlayer
vida = 'Videos/testvids/6.mov'
vidb = 'Videos/testvids/3.mov'
vidc = 'Videos/testvids/t2.mp4'
vidd = 'Videos/testvids/t3.mp4'
default = 'Videos/testvids/t1.mp4'

from keyb import KBHit

player = OMXPlayer(default, args=['--no-osd','--blank'],)

try:
      # KEYBOARD control
      kb = KBHit()
      while True:
          if kb.kbhit():
              c = kb.getch()
              print(c)
              if c == chr(27) or c == 'q': # ESC or 'q' key to quit
                  break
              elif c == '1':
                  player.load(default)
                  player.play()
              elif c == '2':
                  player.load(vida)
                  player.play()
              elif c == '3':
                  player.load(vidb)
                  player.play()
              elif c == '4':
                  player.load(vidc)
                  player.play()
              elif c == '5':
                  player.load(vidd)
                  player.play()
                  
          sleep(1)

except KeyboardInterrupt:
      print('KeyboardInterrupt')

# Kill the `omxplayer` process gracefully.
player.quit()
```

The keyboard KBHit() functionality in [keyb.py.zip](https://github.com/willprice/python-omxplayer-wrapper/files/1069274/keyb.py.zip).

Fix in python-omxplayer-wrapper:: player.py
```
....

    def _setup_dbus_connection(self, Connection, bus_address_finder):
        logger.debug('Trying to connect to OMXPlayer via DBus')
        self.tries = 0 #fix issue-85
        while self.tries < 50:
            logger.debug('DBus connect attempt: {}'.format(self.tries))
            try:
....
```

closed 2017-06-12 by willprice

JE-Random-Tech, 2019-09-22:
Hi,
this answer is a little late but rather late then never ;-). Regarding point 1 you could increase the GPU memory and try if the videos can be played with the terminal (omxplayer pathofvideo). For the second point i made a video which you can see [here](https://www.youtube.com/watch?v=a476_6VVs5Y). This also includes how to set the Gpu memory.
Hope this helps .

commit d3e3b1be (2017-06-12): Fix #85

Previously we never reset `tries` after loading a new file, and since the
variable was an instance variable it would accumulate the total number of tries
to load any file, this caused issues when opening multiple files as these would
eventually reach the retry threshold and no new files would be able to be loaded.

See https://github.com/willprice/python-omxplayer-wrapper/issues/85 for discussion

Thanks to @jehutting for the bug hunting and fix.
commit 8ffe9d64 (2017-06-12): Fix #85

Previously we never reset `tries` after loading a new file, and since the
variable was an instance variable it would accumulate the total number of tries
to load any file, this caused issues when opening multiple files as these would
eventually reach the retry threshold and no new files would be able to be loaded.

See https://github.com/willprice/python-omxplayer-wrapper/issues/85 for discussion

Thanks to @jehutting for the bug hunting and fix.