willprice/python-omxplayer-wrapper #86: Video playback fails with the exception dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply.

jatinkrmalik, 2017-06-08:
# Issue Report
## Description
Video playback fails with the exception **dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply**. I have read almost all the issues and also checked forum but all the other people encountering the same error have either done a mistake in the args=[] or trying to invoke multiple player instances without closing the old ones. 

## Problem reproduction
Here is the code executed (pretty basic) and the log stack of the issue. I have take steps like rebooting the pi and reinstalling the dependencies but it is still failing. 

Steps to reproduce:
1. Save code pastebin below to test.py
2. Execute `python test.py` 

- Code executed: https://pastebin.com/qvgWv8Cq
- Log stack: https://pastebin.com/v88KZg5K

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.2` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0-2` |
| `python` (`python --version`) | `3.4.2` |
| `omxplayer` (`omxplayer --version`) | `dfea8c9` |

-------

# Feature Request
## Description
Include either instructions to install python**3**-dbus or package it in this project itself.

jehutting, 2017-06-11:
Hello @jatinkrmalik 
When you run into this issue, can you post the output of `ls -l /tmp/omxplayerdbus*`? Also post the output of `whoami`. Thanks in advance!
I think the problem is caused by the `BusFinder::find_address_file()` picking up the wrong D-Bus bus address. **After** you have posted the requested info, you can remove the `/tmp/omxplayerdbus*` files (as these are re-created upon omxplayer startup). Hopefully the error is gone.

willprice, 2017-06-12:
1. How long is your media file?
2. Agree with @jehutting, this is probably due to picking up the incorrect bus address/stale bus address files. Can you post the requested information

jatinkrmalik, 2017-07-11:
Hi @willprice , It weird that it started working on a fresh install of Raspbian. I feel dbus got corrupted or similar.

willprice, 2017-07-11:
If you do see the problem again, post the output of the `omxplayerdbus*` files in `/tmp` then we can figure out whether you've got stale files present or not causing issues; this has been the source of issues for me previously.

Closing for the time being until we can reproduce.

closed 2017-07-11 by willprice