willprice/python-omxplayer-wrapper #88: problem with omxplayer quit, load function - cant play multiple files

michauPe, 2017-07-07:
# Issue Report
## Description
I have rpi3 and appIication in python2. I can play and controll video with first use of my application, but after loading new file or quiting omxplayer ( by quit() function) in code, new wideo cant be played. I can see only black blinking screen.

I was digging around this problem for days, but i cant solve this - I think this is some problem with dbus. HELP !:)

## Problem reproduction
Just load next file in code

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.1.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0` |
| `python` (`python --version`) | `2.7.9` |
| `omxplayer` (`omxplayer --version`) | `6c90c75` |

-------

# Feature Request
## Description
A short description (preferably a one liner) of the feature you desire.

## Example use case
An example of how the feature you request would be used, demonstrating it's
utility (and why the current solution doesn't solve your problem).

willprice, 2017-07-08:
Cab you provide the file you reference please.

michauPe, 2017-08-04:
reference file is simple, the demo one from smoke_tests

I have noticed that problem is with 1080p resolution. With 720p i can play, quit, load several times. With 1080p i cannot quit and load new video...

jehutting, 2017-08-05:
Try to increase your `gpu_mem` in /boot/config.txt.

willprice, 2017-09-26:
Not been able to reproduce, please comment with a minimal working example and more details on your setup if you want me to look into it

* Raspberry pi model
* Contents of `config.txt`
* Contents of `cmdline.txt`

closed 2017-09-26 by willprice