willprice/python-omxplayer-wrapper #90: Videos play before being called and cannot restart second video

JonnyAlpha, 2017-07-27:
Will, I am trying to build a visual front end for a robot. When nothing is happening a video should loop and when motion is detected a second video starts. 
**Issue:** Both videos in my script play automatically before they are called. Once they have both completed the first video which is on a loop carries on ok but when trying to play the second video it will not play and just remains frozen for the duration. Once it has finished playing (frozen) the loop video continues.  If I make both videos loop they both just keep playing?

Here is my code. First video should loop and when motion is detected it should pause and the second video should play.

`import os
import sys
from time import sleep
from omxplayer import OMXPlayer
from gpiozero import MotionSensor
from signal import pause
from picamera import PiCamera

pir = MotionSensor(23)

camera = PiCamera()
vid1 = OMXPlayer('/home/pi/Videos/Vid_Wait_Loop1.mp4',args=['--win', '100 100 640 480','--loop'])
vid2 = OMXPlayer('/home/pi/Videos/Vid_Name_Hammerstein.mp4',args=['--win', '100 100 640 480'])

def main():
    initiate()
    while True:
        print("playing vid1")
        vid1.play()
        if pir.motion_detected:
            motion_detected()
        else:
            no_motion()
    except KeyboardInterrupt:
        print("terminated by user")
        vid1.quit()
        vid2.quit()
        camera.close()
            try:
                sys.exit(0)
            except SystemExit:
                os.exit(0)
    finally:
           vid1.quit()
           vid2.quit()
         camera.close()
     print("bye bye")

def initiate():
    print("CENTRAL AI Startup - Running initial setup")
    sleep(1)
    print("Starting Central AI visual front end")
    sleep(1)
    print("Motion detection activated")
    sleep(1)
    print("Security system activated")

def no_motion():
    print("All quiet")
    vid1.play()

def motion_detected():
    print("Intruder Detected")
    sleep(2)
    print("Pausing loop")
    vid1.pause()
    print("Playing Vid_Name_Hammerstein")
    vid2.play()
    sleep(10)
    vid2.pause()`

Rasbian Jessie updated and upgraded
python-omxplayer-wrapper downloaded and installed yesterday.

jehutting, 2017-07-30:
Hi @JonnyAlpha 
Aha...Mr. Hammerstein... that's a long time ago I saw his videos; can't remember when or where :-)

_Currently_ Will's wrapper isn't able to play multiple omxplayers. It will be when @willprice pulls in @longagofaraway pull-request #89.

The pull request surprised Will on the wrapper not initially pausing the video :-) So you need to add the ```pause=True``` argument for the time being.

Also it is good to realize that once the playing of vid2 is finished, there is no vid2 (real) omxplayer any more. Sending commands to vid2 will fail!

So I guess it is better to do in the ```motion_detected``` function the creation of vid2 and use ```play_sync``` to run video2 until it is completed, and continue with resuming the video1 playing.  You can start video1 in the ```no_motion``` function, but until the PIR gives the event  ```when_no_motion``` the video1 stays in the paused state -a freezing video1-.

PIR events... I can only agree on MrYsLab's comment on the Raspberry Pi Forum. I have a PIR so I can have a look at that too.

Attached my version of your code. The PIR is simulated by the keyboard; pressing the 'm' -key will simulate the ```when_motion_detected``` (and the 'n'-key the ```when_no_motion``` event, but that one I don't use). Use the 'ESC' or 'q'-key to quit the program. 
Again it only works with longagofaraway pull request.

[JonnyAlpha-Mr-Hammerstein-20170930A.zip](https://github.com/willprice/python-omxplayer-wrapper/files/1185511/JonnyAlpha-Mr-Hammerstein-20170930A.zip); can add the videos for other users if you allow me to do so.

I made a small video of the running program which I can post to you if you are interested.

JonnyAlpha, 2017-08-18:
My apologies to both of you. I saw these responses but have just not had the time to look at anything until today :-(
I have just installed the lastest python-omxplayer-wrapper and will be having a play. 
@jehutting thanks for taking the time to look at my code. I didn't realise I needed so many amendments - or the addition of a dbus name. 
Here is the link to some of the videos I have made that I am trying to use in my application:
https://www.dropbox.com/sh/kbbsta5l9mnx0so/AACBJD6bibq38bgASUHwOLF6a?dl=0

Can we continue to discuss on here or should we deal with this via email separately?

I have just run your code and the Vid_Wait_Loop is playing (although I am running the Pi Headless in a Hotel so can only hear the sound via a plugged in speaker). But the terminal then froze with the following message:

`Traceback (most recent call last):
  File "jonnyalpha.py", line 26, in <module>
    vid1 = OMXPlayer(videos+'/Videos/Vid_Wait_Loop.mp4',args=['--no-osd', '--no-keys', '--win', '100 100 640 480', '--loop', '-o', adev], dbus_name='org.mpris.MediaPlayer2.omxplayer1', pause=True)
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 82, in __init__
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 166, in load
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 89, in _load_source
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 129, in _setup_dbus_connection
  File "build/bdist.linux-armv7l/egg/omxplayer/dbus_connection.py", line 24, in __init__
  File "build/bdist.linux-armv7l/egg/omxplayer/dbus_connection.py", line 28, in _create_connection
  File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 122, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.`

I did download and install dbus python3 as well as already having dbus installed?

JonnyAlpha, 2017-08-18:
Oh and yes Hammerstein was one of the ABC Warrior Robots from the comic 2000AD. Also appeared in the 1995 version of 'Judge Dredd' with Sylvester Stalone.
I am building a 'virtual' robot with a visual front end.

jehutting, 2017-08-18:
@JonnyAlpha So I assume it is working as in the notification I got you are saying you had a frozen window?

Didn't now about Judge Dredd. I looked it up, but my Mr. Hammerstein videos are dated '27-januari-2015'.

We can continue discussing here -as I got a notification of posted messages, or by mail; whatever you like.

JonnyAlpha, 2017-08-19:
@jehutting Yes it is working fine, I amended the post but initially I reported that the terminal froze after running the program but the second time it worked fine.
The videos were made in 2015 when I first came up with the idea.

willprice, 2017-09-26:
If you can provide an  example that I can run without having to build a robot I'll have a look into it, but until then I'll close this as I cannot reproduce.

closed 2017-09-26 by willprice