willprice/python-omxplayer-wrapper #92: "ImportError: No Module named 'dbus'

chelseamw13, 2017-08-23:
Delete above and including this line.

# Issue Report
## Description
I am able to run the wrapper using python2, however when I run it with python3 (after downloading your python_2_3 support) I get the above error. I have re installed dbus quite a few times but I believe the problem is that it's not installing properly. Do you have a recommended way to install dbus?

## Problem reproduction
A **MINIMAL** example of how to reproduce the example, don't post the contents
of your whole file. Take the time to trim it down to the minimal number of steps
required

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.1.0` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0-2` |
| `python` (`python --version`) | `2.7.9` |
| `omxplayer` (`omxplayer --version`) | `dfeaBc9` |

-------

# Feature Request
## Description
A short description (preferably a one liner) of the feature you desire.

## Example use case
An example of how the feature you request would be used, demonstrating it's
utility (and why the current solution doesn't solve your problem).

willprice, 2017-08-23:
Try off master, I thought I merged py3 support in.

chelseamw13, 2017-08-24:
I tried it of master as well but no luck!

chelseamw13, 2017-08-24:
Thank you anyways! I fixed it by creating a virtualenv -- was having problems with python3.4 vs python3.5

closed 2017-08-24 by chelseamw13