willprice/python-omxplayer-wrapper #94: DBus.Error.NoReply while is_playing checking

unuaunco, 2017-09-13:
# Issue Report
## Description
Hello. I am trying to build a simple office TV player with wrapper. But  script returns DBusException error. Script crushes, while checking .is_playing (see example below).
Is there any ways to run code like this?
Thank you!
## Problem reproduction
This code:
```
#!/usr/bin/python
from omxplayer import OMXPlayer
from time import sleep

file_path_or_url =  '/home/pi/21.mp4'

player = OMXPlayer(file_path_or_url, args=['-I','-r','--vol','-6000','--nodeinterlace'])

player.pause()
sleep(5)
player.play()
while(player.is_playing()):
 sleep(1)

player.quit()
```
And this error:
```
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/21.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42mp41
    creation_time   : 2017-05-22T11:31:54.000000Z
  Duration: 00:00:10.05, start: 0.000000, bitrate: 17671 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, bt709), 1920x1080 [SAR 1:1 DAR 16:9], 17156 kb/s, 24 fps, 24 tbr, 24k tbn, 48 tbc (default)
    Metadata:
      creation_time   : 2017-05-22T11:31:55.000000Z
      handler_name    : Alias Data Handler
      encoder         : AVC Coding
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 317 kb/s (default)
    Metadata:
      creation_time   : 2017-05-22T11:31:55.000000Z
      handler_name    : Alias Data Handler
Traceback (most recent call last):
  File "./omx_helloWorld.py", line 15, in <module>
    while(player.is_playing()):
  File "<decorator-gen-90>", line 2, in is_playing
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 45, in wrapped
  File "<decorator-gen-89>", line 2, in is_playing
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 53, in wrapped
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 665, in is_playing
  File "<decorator-gen-28>", line 2, in playback_status
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 45, in wrapped
  File "<decorator-gen-27>", line 2, in playback_status
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 53, in wrapped
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 307, in playback_status
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 738, in _player_interface_property
  File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 732, in _interface_property
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
```

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.1 (stretch)
Release:        9.1
Codename:       stretch

```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.3` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.0` |
| `python` (`python --version`) | `2.7.13` |
| `omxplayer` (`omxplayer --version`) | `5a25a57` |

willprice, 2017-09-26:
That's because the player has completed playing and the process has exited so all further dbus method calls will fail; this should be addressed inside the library by returning last queried state or similar.

willprice, 2017-09-26:
You should use a `try/except` block to catch the DBusException as it implies the player has exited.

himijendrix24, 2018-11-12:
Are there any examples on how to use the dBus finder? Does it even have to be used manually? 
Using try/except causes the problem for me, that many things (maximum play length, volume ...) which I want to change dynamically have to be extra checked and logged, for the case the dbus connection is lost.

willprice, 2018-11-13:
Nope, you shouldn't have to use it. If you're getting DBusExceptions then it's likely omxplayer has died. Have you tried clearing out any `/tmp/omxplayer*` files? If so, does this help with your issue?

closed 2019-10-09 by unuaunco

veppan, 2021-01-11:
how did you solve the problem in the end?
a similar problem.

veppan, 2021-01-11:
> # Issue Report
> ## Description
> Hello. I am trying to build a simple office TV player using wrapper. But script returns DBusException error. Script crushes, while checking .is_playing (see example below).
> Is there any ways to run code like this?
> Thank you!
> 
> ## Problem reproduction
> This code:
> 
> ```python
> #!/usr/bin/python
> from omxplayer import OMXPlayer
> from time import sleep
> 
> file_path_or_url =  '/home/pi/21.mp4'
> 
> player = OMXPlayer(file_path_or_url, args=['-I','-r','--vol','-6000','--nodeinterlace'])
> 
> player.pause()
> sleep(5)
> player.play()
> while(player.is_playing()):
>  sleep(1)
> 
> player.quit()
> ```
> 
> And this error:
> 
> ```
> Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/pi/21.mp4':
>   Metadata:
>     major_brand     : mp42
>     minor_version   : 0
>     compatible_brands: mp42mp41
>     creation_time   : 2017-05-22T11:31:54.000000Z
>   Duration: 00:00:10.05, start: 0.000000, bitrate: 17671 kb/s
>     Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, bt709), 1920x1080 [SAR 1:1 DAR 16:9], 17156 kb/s, 24 fps, 24 tbr, 24k tbn, 48 tbc (default)
>     Metadata:
>       creation_time   : 2017-05-22T11:31:55.000000Z
>       handler_name    : Alias Data Handler
>       encoder         : AVC Coding
>     Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 317 kb/s (default)
>     Metadata:
>       creation_time   : 2017-05-22T11:31:55.000000Z
>       handler_name    : Alias Data Handler
> Traceback (most recent call last):
>   File "./omx_helloWorld.py", line 15, in <module>
>     while(player.is_playing()):
>   File "<decorator-gen-90>", line 2, in is_playing
>   File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 45, in wrapped
>   File "<decorator-gen-89>", line 2, in is_playing
>   File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 53, in wrapped
>   File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 665, in is_playing
>   File "<decorator-gen-28>", line 2, in playback_status
>   File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 45, in wrapped
>   File "<decorator-gen-27>", line 2, in playback_status
>   File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 53, in wrapped
>   File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 307, in playback_status
>   File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 738, in _player_interface_property
>   File "build/bdist.linux-armv7l/egg/omxplayer/player.py", line 732, in _interface_property
>   File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
>     **keywords)
>   File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
>     message, timeout)
> dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
> ```
> 
> ## Environment details
> * OS (`lsb_release -a`):
> 
> ```
> Distributor ID: Raspbian
> Description:    Raspbian GNU/Linux 9.1 (stretch)
> Release:        9.1
> Codename:       stretch
> ```
> 
> Software	Version
> `python-omxplayer-wrapper`	`0.2.3`
> `python-dbus` (`dpkg -s python-dbus`)	`1.2.0`
> `python` (`python --version`)	`2.7.13`
> `omxplayer` (`omxplayer --version`)	`5a25a57`

how did you solve the problem in the end?
i have a similar problem.

mghanei, 2021-01-29:
I have the same problem here. Could you please tell, how you solved it?
Thanks!

mghanei, 2021-01-29:
> how did you solve the problem in the end?
> i have a similar problem.

Use this:

```
try:
  player = OMXPlayer(file_path_or_url, args=['-I','-r','--vol','-6000','--nodeinterlace'])
  
  player.pause()
  sleep(5)
  player.play()
  while(player.is_playing()):
   sleep(1)
except:
  player.quit()
```