willprice/python-omxplayer-wrapper #100: Request callback example

chrispo-m, 2017-09-19:
# Feature Request
## Description

Provide an example or two on how to use the stopEvent, playEvent, etc, callback features

## Example use case

I'm attempting to make use of the callback features of 0.2.3 but obviously doing something wrong.  Could you all provide an example or two, please?

My use case

`import sys
import time
import omxplayer.player
import omxplayer.keys

def player_stop():
	print ("player stopped callback")

def player_pause():
	print ("player pause callback")

def player_start():
	print ("player start callback")

# main
# ---------
# - http://python-omxplayer-wrapper.readthedocs.io/en/latest/omxplayer/

print ("loading")
player = omxplayer.player.OMXPlayer('/home/clm/music/Legend/14 Jamming.mp3')

player.stopEvent = player_stop()
player.playEvent = player_start()
player.pauseEvent = player_pause()

... etc
`
The callbacks all fire when set in the code above, but don't fire on the actual start/stop/pause events.

Thanks!

willprice, 2017-09-19:
You're invoking your callbacks when assigning them, try

```python

player.stopEvent = player_stop
player.playEvent = player_start
player.pauseEvent = player_pause
```

chrispo-m, 2017-09-19:
Thanks much, Will.  Jack of all programming languages, master of none I am ...

If I can ask another question about your api / callback structure -- what's the best way to detect end of playback?  The stopEvent callback didn't fire when my mp3 finished.  I'm looking for a signal, other than polling is_playing() or playback_status() to know when the current file is done playing and to load the next.

Thanks for all of your work on this api;  it's quite useful.

--Chris

willprice, 2017-09-19:
Hi Chris,

Sadly I don't currently have a way of doing that, your only option is to pol `is_playing` or use the `play_sync` method which blocks for you (by polling status).

What's the exact use case? You have a bunch of MP3s, do you know them all at the start of runtime or are they dynamically collected? Should you always play the next MP3 after the last? etc

chrispo-m, 2017-09-19:
Hi Will,

Yep, that's pretty much what Id like to do.  I'm attempting to build a Raspberry Pi based audio player.  It'll either point toward a directory (album) of mp3's on my NAS, or toward a .pls file from an internet radio station (once I figure out how that works...)

My questions stem from the need to know when one file's done playing so I can call a player.load(next file)  for the next.  I could do some intelligent polling of is_playing using the length from player.duration(), though...

And a second noobie question -- any trick to getting the metadata dict?    

File "audio_omx.py", line 52, in <module>
    print ("metadata " + player.metadata())
AttributeError: 'OMXPlayer' object has no attribute 'metadata'

Kind regards,
Chris

willprice, 2017-09-19:
Hi Crhis,

The `metadata` support is only on master and not in the released version on PyPI (yet.) you can install from the `master` branch by running `pip install git+git://github.com/willprice/python-omxplayer-wrapper.git@master`.

Your application sounds quite typical, why not just use [`play_sync`](http://python-omxplayer-wrapper.readthedocs.io/en/latest/omxplayer/#omxplayer.player.OMXPlayer.play_sync) method?

chrispo-m, 2017-09-19:
Thanks, Will,

I should also add that I'm going to use Tornado as a browser-based gui for this.  I have a tick-timer of sorts set up now to avoid blocking the web services.  So, I'm not sure if I'd hose this up if I blocked via play_sync.  But I could use this tick to check is_playing.

Thanks again for all your help and for most-useful omxplayer wrapper, 

--Chris

willprice, 2017-09-19:
Actually, I'm going to revise that previous suggestion RE event, you need to use `+=` to append the event handler, `=` will just overwrite the event object.

```
player.stopEvent += player_stop
player.playEvent += player_start
player.pauseEvent += player_pause
```

willprice, 2017-09-19:
Tornado has its own main loop right? I should really have a look at allowing people to inject their own main loops in... I'm thinking of switching away from the `dbus-python` library anyway as even the freedesktop guys [list it as deprecated](https://www.freedesktop.org/wiki/Software/DBusBindings/) now in favour of other third party libs. I haven't had the time nor inclination to check them out yet. I'd like to refactor the unit tests first to get into a state where moving dbus implementation would be easier.

willprice, 2017-09-27:
The advanced example on the front page of the docs (http://python-omxplayer-wrapper.readthedocs.io/en/latest/) shows how to use the event mechanism.

closed 2017-09-27 by willprice