willprice/python-omxplayer-wrapper #102: Can stream but not load local files

kirkkeller, 2017-09-21:
# Issue Report
## Description
When I run the example Python script for URI streaming, the script performs as expected. However, if I try to run a locally stored mp4 file or mov as a file, I get an error.

Trace of the script shows something amiss with Path:

 File "/usr/lib/python3.5/urllib/parse.py", line 99, in <genexpr>
	return tuple(x.decode(encoding, errors) if x else '' for x in args)
AttributeError: 'PosixPath' object has no attribute 'decode'

I'm at a bit of a loss as to how to proceed. Am I missing something or where in general should I be looking for a solution?

I hope my question isn't too vague. I just didn't want to troubleshoot in the wrong direction.

## Problem reproduction

#!/usr/bin/env python3
from omxplayer.player import OMXPlayer
from pathlib import Path
VideoPath = Path("/home/pi/Videos/HappyKittens.mp4")
player = OMXPlayer(VideoPath)
player.play()
player.quit()

## Environment details
* OS (`lsb_release -a`):
```
Distributor ID: Raspbian
Description:	Raspbian GNU/Linux 9.1 (stretch)
Release:    	9.1
Codename:   	stretch
```

| Software | Version |
|----------|---------|
| `python-omxplayer-wrapper` | `0.2.3` |
| `python-dbus` (`dpkg -s python-dbus`) |  `1.2.4-1` |
| `python` (`python --version`) | `2.7.13` |
| `omxplayer` (`omxplayer --version`) | `5a25a57 (debian)` |

willprice, 2017-09-21:
Hi @kirkkeller,

This is definitely a bug on my side, sorry for the trouble, drop using `Path` for the time being and just pass in a string until I fix this. Currently it seems `Path` is misinterpreted as a URL hence the errors about a `decode` method not being found.

Thanks for the report! I'll notify you when this is fixed.

kirkkeller, 2017-09-21:
That did work, thanks. However, I did have to add a short sleep between instantiating the OMXPlayer and player.play(). Otherwise, I'd get a video time display but no video. Very cool wrapper, though. Can't wait to start using it tonight.

willprice, 2017-09-21:
Hi @kirkkeller,
Glad that fixed the issue, please do give me feedback on how you find the experience. Irritatingly omxplayer doesn't provide much information on when the player is ready to start receiving events after being fully initialised so its a bit of a guessing game whether the player is ready or not after spawning the `omxplayer` process.

commit 1d9119c6 (2017-09-26): Merge pull request #104 from willprice/develop

Fix #102
closed 2017-09-26 by willprice