willprice/python-omxplayer-wrapper #103: No module named 'omxplayer' after review  #66 & #77

deepline1986, 2017-09-25:
I run the example code by python3. The error code is "No module named 'omxplayer'"

I've checked the #66 & #77 but it seems cannot solve the issue.

The example code is coming from "http://python-omxplayer-wrapper.readthedocs.io/en/latest/" and it seems support python3 already.

Thanks for the support!

The example code is

-----------------------------------------------------------------------------
#!/usr/bin/env python3

from omxplayer.player import OMXPlayer
from pathlib import Path
from time import sleep

VIDEO_PATH = Path("../home/pi/Videos/1080p.mkv")

player = OMXPlayer(VIDEO_PATH)
player.play()

sleep(5)

player.quit()
-----------------------------------------------------------------------------

willprice, 2017-09-26:
Can you provide more information:
* How did you install the library
* What version of python 3 you are using.

StevenJUlbrich, 2017-11-04:
Hello 
I am having the same issue 
Python 3.5.I default Jan 19, 2017

I followed the install instructions from http://python-omxplayer-wrapper.readthedocs.io/en/latest/

the first line would not resolve properly but a basic check showed that libdbus-1.3 present
the pip install was executed 
rebooted the raspberry pi

RESTART: /home/ip/python-omxplayer-wrapper/examples/video_file.py
Traceback(most recent call lst):
File "/home/pi/python-omxplayer-wrapper/examples/video_file.py" line 3, in <module> from omxplayer.player import OMXPlayer: No module named 'omxplayer'

jehutting, 2017-11-05:
Maybe if you use pip**3** (in stead of pip)? pip3 is normally the command to use for installing packages for Python3?
```
$ pip3 install omxplayer-wrapper
```

StevenJUlbrich, 2017-11-09:
pip3 did the trick

RonDraper, 2018-05-21:
I'm basically having the same issue.  I have done the pip3 install omxplay-wrapper command.  Everything is up to date and requirements already satisfied.
My issue is that my import works while using the terminal and why using Thonny to execute the py module.  But when I try to get the raspberry pi to execute it on startup it fails.

willprice, 2018-05-21:
What is thorny, how are you trying to start it up and can you capture a stack trace of the issue? I'm guessing you're invoking it with the wrong version of python and hence a different suite of site-packages where `omxplayer-wrapper` isn't installed

closed 2018-11-03 by willprice

yoyojacky, 2021-05-25:
i have same issure too:
```
pi@RPi8g:~/Videos $ pip3 install omxplayer-wrapper
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: omxplayer-wrapper in /home/pi/.local/lib/python3.7/site-packages (0.3.3)
Requirement already satisfied: evento in /home/pi/.local/lib/python3.7/site-packages (from omxplayer-wrapper) (1.0.2)
Requirement already satisfied: decorator in /usr/lib/python3/dist-packages (from omxplayer-wrapper) (4.3.0)
Requirement already satisfied: dbus-python in /home/pi/.local/lib/python3.7/site-packages (from omxplayer-wrapper) (1.2.16)
Requirement already satisfied: pathlib2 in /home/pi/.local/lib/python3.7/site-packages (from omxplayer-wrapper) (2.3.5)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from pathlib2->omxplayer-wrapper) (1.12.0)
pi@RPi8g:~/Videos $ python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import omxplayer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/Videos/omxplayer.py", line 2, in <module>
    from omxplayer.player import OMXPlayer
ModuleNotFoundError: No module named 'omxplayer.player'; 'omxplayer' is not a package
>>> 
>>> 
>>> import OMXPlayer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'OMXPlayer'
>>> 
>>> 
>>> 
>>> quit()
```

Gillissie, 2022-01-28:
Why is this issue closed?