willprice/python-omxplayer-wrapper #114: Small Issue with set position

therysin, 2017-10-16:
Hey Will, thanks for all the work on this library. I managed to get this issue on both a raspi 2 model b and 3 model b. 

Sometimes when I use the player.set_position() command it's a few seconds off. 

To reproduce the problem I did a simple video playback with a video that had timestamps onscreen, and using the set position command to go to 20 seconds, 25, 45 and repeat. I did a short video in for you with the issue. https://youtu.be/rNVN8iJ6Hts

I'm really new with the raspberry pi, so I couldn't really follow the template fully. I'm on the latest version of the wrapper (installed from pip 2 days ago) and I only installed raspbian about 2 weeks ago from the official site. 

Thanks for everything, I'm using your wrapper to synchronize video playback for a school project. Thanks again.

jehutting, 2017-10-16:
Is it possible to post the video you are playing?

therysin, 2017-10-16:
Thanks for the reply, here is the video: http://pocketvj.com/video/small_testfile.mp4

jehutting, 2017-10-16:
I played the video with (the real) omxplayer and used dbuscontrol.sh to do the setpositions. I can see the few seconds off too.

AFAIK when omxplayer executes a setposition, it _seeks_ to the nearest _keyframe_. In your case setposition of 20 secs results in a keyframe around 18 secs, 25 secs to keyframe 24 secs, and 45 secs to keyframe 45 secs. That's the time off time you get. 

I think it depends on how the video is made. E.g. take [https://www.edius.de/test/foo.mp4](https://www.edius.de/test/foo.mp4), a h264/1080p50 video of 30 seconds and look how almost accurate the setpositions are.

therysin, 2017-10-16:
Thanks for the reply jehutting. I think I understand now. Can you offer any advice on setting the position accurately for any(sub 40 fps) video though? I was looking at the seek function but that seems to look for a key frame around 5 seconds too. Thanks again.

jehutting, 2017-10-17:
Unfortunately I have no advice on how to improve the seeking/positioning. It has been a while ago I looked at it but ffmpeg -on which omxplayer leans on- is just one step too far for me :-(

therysin, 2017-10-17:
That's okay. Thanks for the help and replies 😊. I managed to do synchronization by skipping a little ahead on one player then pausing for the deviation and that works very well. Now the next step for me would be loading more than one video from a directory. Thanks again Jehutting!

closed 2017-10-17 by therysin